betaflight/betaflight #578: Add support for data flash chip on the DTFc target

barthalamu, 2016-06-24:
I tried installing a flash chip I pulled off a bad Dodo FC and it's not recognized in the configurator, it only shows blackbox as being available for serial logging. I noticed it wasn't defined in the DOGE target.h file. Any particular reason?
Could someone please enable data flash on the DTFc? Or at least tell me how to change the target file to add it and compile myself? I'm not a good enough coder to know how to fix it myself. :/

blckmn, 2016-06-25:
It is in the target.h for v3

```
#define USE_FLASHFS
#define USE_FLASH_M25P16
#define M25P16_SPI_SHARED
#define M25P16_CS_PIN       PC15
#define M25P16_SPI_INSTANCE SPI2
```

Can you compile it? Otherwise I can send you a HEX.

mikeller, 2016-06-25:
This could help you to get set up to build with your own modifications: https://hub.docker.com/r/mikeller/betaflight-build/

barthalamu, 2016-06-25:
Cool! Thanks!
Yeah, I can compile (Followed the build instructions for Mac), but embedded stuff is over my head.  ;)
Also wasn't sure which pin was being used for flash, since I noticed other targets used GPIO12, but the DTFc uses that for the baro..

blckmn, 2016-06-26:
@barthalamu did you manage to get this one resolved?

borisbstyle, 2016-06-26:
I will close this one guys to prevent cluttering

closed 2016-06-26 by borisbstyle