betaflight/betaflight #582: Custom Hex file flashing issue

gaelj, 2016-06-25:
First of all I apologize for opening an issue for this, but this problem is stalling my work on the LED strip port and I cannot find what is wrong.
I'm using Eclipse and Cygwin in windows 7 and SPR F3 and SPR F3 EVO boards.
I have absolutely no problem compiling and flashing both Cleanflight and iNav to both boards, be it from Eclipse or by running the fake travis build script in Cygwin.
Betaflight on the other hand, I can compile, but after I flash (via Betaflight configurator), when the board restarts Windows complains it did not recognize the USB device. Both boards show the same behaviour. The boards is running though : the red LED starts blinking when I tlit it.
I have tried compiling 2.8.1 tag, current Betaflight branch, development branch. All have the same result.
The only way I can successfully flash Betaflight is by selecting an official online hex in the configurator.

If anyone could point me out anything I could be doing wrong I'd be reallly grateful (and able to start testing the LED strip port !).

TheAngularity, 2016-06-25:
I had the same problem the VCP Port was not working. 
Now i'm using cygwin and a toolchain not greater as 4.9 and the custom hex is working very well.

gaelj, 2016-06-25:
@TheAngularity, thank you very much. Using Toolchain 4.9 fixed it.

MJ666, 2016-06-25:
Makeing the change from this PR https://github.com/cleanflight/cleanflight/pull/2120 will also make it work with latest compiler.

gaelj, 2016-06-25:
It would be nice to fix this. Re-opening the issue :)

MJ666, 2016-06-25:
Have merged the related PR.

closed 2016-06-25 by gaelj