betaflight/betaflight #584: roll_yaw_cam_mix_degrees causing odd roll/pitch

carlylea, 2016-06-25:
I have experienced an issue where the roll and pitch rates are different when adding in roll_yaw_cam_mix_degrees.  When I set up my quad using 0 degrees of mix, the quad will roll and pitch at virtually the same speed, but once I add roll mix, the quad will roll much slower (at least half the roll rate with 35 degrees of mix).  Is this something I just have to deal with in my roll rate number?  Thanks for all your hard work Boris.

borisbstyle, 2016-06-25:
Well you mix them together. Of course that rate will change as roll means
yaw + roll.

FieserKiller, 2016-06-25:
but, is that not a bug? 
In my opinion roll/yaw rates should be calculated after roll_yaw_cam_mix_degrees was applied.

borisbstyle, 2016-06-26:
No its not a big. You choose to mix the 2 axis. If the rates are not the same. Lets say yaw is slow rate and you mix it into roll than roll will be slowed down too

closed 2016-06-26 by borisbstyle