betaflight/betaflight #596: 2.9.0spracingf3.hex.CLI  SET airmode-activatt-throttle=10310

E120103016, 2016-06-26:
今日使用BF2.9.0程式發現一個很奇怪現象.可能是個BUG.在CLI裡面SET airmode-activatt-throttle=10310.理論上可以調整數據１０００－２０００．不可能會產生１０３１０．把它修改為１３５０以後．如果你在１．３版配置器上ＰＩＤ再一次設定保存（按ｓａｖｅ）．再回ＣＬＩ裡你會發現又恢復為SET airmode-activatt-throttle=10310．所以我相信不久ＢＦ又會重新改程式．

KiteAnton, 2016-06-26:
Can you please take this in English instead?

martinbudden, 2016-06-26:
Google translate says:

> Today, the use of BF2.9.0 program found a very strange phenomenon may be a BUG. In the CLI inside SET airmode-activatt-throttle = 10310. Theoretically you can adjust the 1000-2000 data. Unlikely to produce 10310. It later revised to 1350. PID again if you save settings (press save) on the Configurator Version 1.3. Where you will find a return to CLI and return to the SET airmode-activatt-throttle = 10310. Therefore, I believe that soon BF will again change the program.

martinbudden, 2016-06-26:
Closed, replaced by issue https://github.com/betaflight/betaflight/issues/598

closed 2016-06-26 by martinbudden