betaflight/betaflight #597: BF 3.0.0 serial protocol issue: Can't USB connect anymore

RipperDrone, 2016-06-26:
Tried 3.0.0-358 on my SPRACINGF3 - there still s/t wrong with the serial protocol. Same happened with a previous development version.

I have the serial port configured as usual 115k/COM5 - doesn't connect anymore. I'll have to flash back to 2.9 for now.

blckmn, 2016-06-26:
Does anybody else have this error? @AndersHoglund you have a SPRACINGF3 yes?

RipperDrone, 2016-06-26:
I have 2 quads, both SPRACINGF3, same behavior on both. Went back to 2.8.2 with them...

gaelj, 2016-06-26:
No problem on an EVO.

AndersHoglund, 2016-06-26:
Yes I have a few SPRF3 clones running 3.0.0. builds 358 and 359. No USB connection problem.

AndersHoglund, 2016-06-26:
@RipperDrone , are you using latest BFC 1.4 ? If not, try it.

RipperDrone, 2016-06-26:
@AndersHoglund: Yes, using BFC 1.4. Both SP3s don't connect anymore.
Timing out without making the connection. Tried all sequences of plugging USB in, restarting BFC 1.4, different options in no reboot sequence, flash on connect, manual baud rate (default = 256k), full chip erase. No way. Was perfectly okay in all firmware versions thru 2.9.

borisbstyle, 2016-06-26:
I can connect all of them without issues. Maybe you were using too new compiler

RipperDrone, 2016-06-26:
@borisbstyle: I didn't compile it myself, downloaded from Jenkins build - don't know their compiler version. Regarding USB connection you don't think anything has been changed? (e.g. I read about a commit dealing with u_int8, u_int32 buffering...)

AndersHoglund, 2016-06-26:
I use exactly the same binaries from the Jenkins builds. 
For the record: gcc/arm-none-eabi/4.9.3 in use.

blckmn, 2016-06-26:
There was a change to uint32_t for buffer counters - but if it was a problem it would affect all targets (it is in core code - not target specific). 

Can you download the latest. Previously there was an issue with VCP being enabled for SPRACINGF3, but it uses a USB-serial chip - VCP has since been disabled - but that was 4-5 days ago.

Is yours a genuine SPRACINGF3? it could be that the FC is crashing on boot, hence the no connection and the requirement for DFU mode to reflash. What else do you have connected - can you describe your setup?

RipperDrone, 2016-06-26:
I have 2 Banggood SPRACINGF3's. They are each connected to a FrSky X4R-SB receiver over serial bus, then there's a buzzer and one telemetry pin (SmartPort) for FrSky X4R simple telemetry.

4-5 days ago might fit, since all versions <3.0.0 have worked properly, and 3.0.0 dev never worked on both SPRF3's.

Yes, the FCs seem to be crashing / stalling on reboot - BFC 1.4 just says connecting... and that's it.

@blckmn What exactly do you mean by downloading the latest?

gaelj, 2016-06-26:
If you tilt the board, does it's red LED start blinking ? If so, the software is running, meaning only the USB is a problem.

RipperDrone, 2016-06-26:
Yes, it does.

RipperDrone, 2016-06-26:
BFC 1.4 says 'serial port successfully opened with ID:6' in the status line, however BFC 1.4 does not switch to the config tabs - just shows the 3 tabs welcome, documentation and flashing. Top right button says 'connecting' and seems to not complete the USB connect process. I can still press 'connecting...' and it will revert back to 'connect', so BFC 1.4 itself is not frozen.

blckmn, 2016-06-26:
@RipperDrone the latest i mean pull the latest code and compile or use the latest hex from jenkins (i.e. this morning).

Do you have a link to the FC on banggood? it is possible there is a pin mapping issue.

RipperDrone, 2016-06-26:
@blckmn I have cloned the altest code and built the 3.0.0 dev myself just 10mins ago and tried again - still doesn't connect.

Here's a link to the FC: http://www.banggood.com/Upgrade-NAZE32-F3-Flight-Controller-Acro-6-DOF-Deluxe-10-DOF-for-Multirotor-Racing-p-1010232.html

I've got the 10DOF version, 2 of them. Pinout so far has been in perfect accordance with the documentation I found on the SPRACINGF3:

http://blog.seidel-philipp.de/sp-racing-f3-flight-controller-pinlayout-und-anschlussplan/

Reviews on BG are extremely good (>340 5 stars, too consistent to be faked...)

gaelj, 2016-06-27:
What version of GCC did you build with ?

gaelj, 2016-06-27:
See https://github.com/iNavFlight/inav/pull/314

blckmn, 2016-06-27:
@RipperDrone can you put this into the Target file:

```
#undef I2C1_OVERCLOCK 
#define I2C1_OVERCLOCK false
```

and rebuild?

blckmn, 2016-06-27:
I have a suspicion that it is the MAG that cannot handle the over clocking on the i2c bus.

AndersHoglund, 2016-06-27:
@RipperDrone , I2C overclocking was introduced on the development branch in Jenkins build #50, Try build #49 and #50 and we will know to some degree that this may be the cause. 
http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Development/changes

AndersHoglund, 2016-06-27:
A common problem with I2C is that clock stretching is not implemented. I had a look in the "driver/bus_i2c_soft.c" file, and sure enough, no clock stretching is implemented. However, this driver is not in use. We use the hardware i2c master. Need to check if there is a known problem with this piece of HW or if it may be a driver/config problem. So far only found this: http://www.eevblog.com/forum/microcontrollers/stm32f4-i2c-master-ignoring-clock-stretching/

Reopening this case as it seems to be a real problem.

AndersHoglund, 2016-06-27:
PR #613 issued. 
@RipperDrone please test the linked SPRF3 firmware with this PR, I2C clock stretch enabled.
https://dl.dropboxusercontent.com/u/11080368/betaflight_3.0.0_SPRACINGF3_PR613.zip

RipperDrone, 2016-06-27:
@AndersHoglund @blckmn You guys are great - I2C clock stretch resolves the problem. Confirmed twice. 

In detail: 
- Flashing from 2.8.2 to 3.0.0 (I2C clock stretched mod version) running as usual
- after flashing, no USB connect (freezes in connecting...)
- closing & restarting BFC 1.4 and cycling USB connection (off/on) 
- now connecting works, several times with no issues
- doing a reflash from here: no problem

Conclusion: Flashing still seems to stall BFC 1.4, but I2C stretch renders the FC completely functional with USB connect.

Thank you for doing all the work of isolating the problem and even supplying a ready-made .hex for me to do the verification. I'm at work quite late these days, so I couldn't reply/test earlier, sorry.

RipperDrone, 2016-06-27:
@AndersHoglund @blckmn Thank you guys for isolating the problem and even supplying a ready-made patched firmware for testing. My apologies I couldn't respond earlier, working late these days...

Result of I2C clock stretch: Part success!

- flashed to this firmware - no problem
- then tried to connect - freezing at connecting... like before
- then closed and restartet BFC 1.4 - USB connect working now! Several times disconnect-connect cycling: No problem!
- then reflashed 3.0.0 - even reflashing working now
- tried to connect again - not successful, even after closing and restarting BFC 1.4
- rebootet computer: now connect working again

Bottom line: No connection robustness but definitely big progress.

What else could I test/build for you to diagnose?

AndersHoglund, 2016-06-27:
Try update BFC to 1.5.

RipperDrone, 2016-06-27:
BFC 1.5 = same, trying reboot computer now...

AndersHoglund, 2016-06-27:
Are there any LEDs flashing when the FC does not connect? Like normal flashes when board is moved, or five fast followed by some long flashes (those are error codes).

RipperDrone, 2016-06-27:
Did a hard reset of my computer. Now the FC showed 5 fast flashes, followed by 4 (?) long flashes, then no flashes anymore. After this sequence, connect was ok.

On the other hand, when the FC showed 5 fast flashes and then nothing else, connect was NOT possible. Does this give you a direction?

RipperDrone, 2016-06-27:
Now I seem to slowly see the pattern:

- upon hard reset, connecting the USB cable triggers 5 fast flashes then slow flashes --> USB connect then is ok. I can also do 20+ disconnect/reconnect cycles no problem.
- once plugging the USB cable out and replugging in: 5 fast flashes only --> no more USB connects possible, even upon closing/restarting BFC 1.5

AndersHoglund, 2016-06-27:
Did you flash with "Full Chip Erase"? 
Error codes are here: https://github.com/betaflight/betaflight/wiki/Frequently-Asked-Questions#why-does-my-flight-controller-blinkbeep-lots-of-times-when-powering-up-
Not sure how to interpret that as this list is "1" -based but the codes in system-h are "0"-based.....

RipperDrone, 2016-06-27:
After next hard reset I tried to watch the LED code indetail: 

Red LED blink code when connecting USB cable: 5 short blinks - 2 long blinks - 3 short blinks - 2 long blinks --> USB connect working

Without even trying to launch BFC 1.5, I then unplugged and replugged the USB cable: 5 short blinks only ---> NO connect

Needs a computer hard reset to get back to normal...

RipperDrone, 2016-06-27:
@AndersHoglund Yes I always flash with full chip erase.

AndersHoglund, 2016-06-27:
What if you power up on battery and BEC, no USB connected ? Same blinking ??

gaelj, 2016-06-27:
@RipperDrone I know this may sound crazy, but, just to eliminate the possibility, could you check you have the problem with different USB cables ?

RipperDrone, 2016-06-27:
One more lead found: My previous analysis on the needs for a computer hard reset seems to be a coincidence only: If I let the FC USB-disconnected just sit there for about a minute and then try to plug in USB again, it always works the 1st time. 

However, when I then unplug and replug USB, it's broken again. Seems there is some capacitance that kills the USB connection? As mentioned before, not any problem with Betaflight versions <3.0.0.

Scratching my head...

RipperDrone, 2016-06-27:
tried 3 different cables and 2 different ports on my computer - all the same...

RipperDrone, 2016-06-27:
Red LED blinking pattern when connecting battery without USB and without Radio Transmitter:

1 long - 3 short - infinite long (not linking to transmitter)

When transmitter is on, blinking pattern is like when connecting successfully to USB (long, short-short-short-long-long-long).

If we need to write further on blinking patterns: Let's use "-" for long, "." for short, so pattern above would be: - ... ---

RipperDrone, 2016-06-27:
When I'm connected and try a CLI dump, interface in BFC 1.5 says packet errors: 1, I2C error: 2

borisbstyle, 2016-06-27:
those errors only happen during boot. no big deal from what I see

AndersHoglund, 2016-06-27:
OK, what if you power up with battery and let the FC startup, and then connect USB and BFC ?
I' lost. Not sure what I am looking for. This test may show that some of the i2c devices onboard may be powered only from the ESC rail, not the USB power. But then, why would 2.9.0 run without problems... 
Puzzled and out of ideas....

RipperDrone, 2016-06-27:
@blckmn Completing your former suggestion to mask the I2C out completely - no remedy...

@RipperDrone can you put this into the target.h file for SPRACINGF3:

#ifdef I2C1_OVERCLOCK 
#undef I2C1_OVERCLOCK 
#define I2C1_OVERCLOCK false
#endif
and rebuild?

RipperDrone, 2016-06-27:
@AndersHoglund Tries to connect battery (ESCs signal low beep, then high beep)

THEN connect USB and fire up BFC 1.5, trying to cnnect - no connect

RipperDrone, 2016-06-27:
If blinking error codes (1 long blink after 5 short blinks) are related to sensors: How can I disable gyro/mag/baro sensors in a build and eliminate those as potential factors?

gaelj, 2016-06-27:
@borisbstyle I dug out my SPRF3 (hobbyking) board, updated my BF and BFC repos to the very latest development versions, compiled the hex in Eclipse as usual and flashed it.
Now when I connect the board by USB (nothing else connected to the board, fresh default erase/flash parameters) the red LED flashes 5 times very fast, then stays dark even if I tilt the board.
The COM port is created and can be opened but I cannot actually connect.
Chrome dev tools log says:
```
Selecting last used port: COM7                    serial_backend.js:48 
Connecting to: COM7                               serial.js:116 
SERIAL: Connection opened with ID: 153, Baud: 115200
10 MSP data request timed-out: 1                  msp.js:1144
```
Neither does PR 619 or the target.h change, applied on latest BF code, does the trick for me.
Also to reflash I have to connect the boot jumper and activate flash on connect.

RipperDrone, 2016-06-27:
2nd path: If quad has to sit USB-unplugged for about 2mins before being able to successfully connecting again: Is there any watchdog timers in the virtual COM port interface / driver which block the com buffer for a given wait time and only release after this?

RipperDrone, 2016-06-27:
@gaelj Congrats, so you can replicate the problem now. Did you try just to let your FC sit USB-uplugged for 2 mins and then try to plug it back in and re-connect? Works for me...

RipperDrone, 2016-06-27:
Here is a new key finding: When soft-switching off MAG, I can unplug/replug USB without any issues! Seems the MAG sensor (readout / timing?) is part of the issue.

gaelj, 2016-06-27:
I don't think I have exaxctly the same problems as you.
Over here I think that my board is frozen, except for the USB / COM port, which I can open. But after that there is no communication.
Whatever I do it always behaves in the same way.

gaelj, 2016-06-27:
BTW mine is the deluxe version as well.

RipperDrone, 2016-06-27:
@gaelj Still believe we are having similar issues: When I plug in (with BARO, ACCEL, MAG ON, I get 5 short blinks, then nothing. Status line in BFC 1.5 says 'serial port ID:xxx, but does not get to the point of logging in and giving the device ID, Can't do any further communication then (button on top right stays in 'connecting...' state).

However, if I let the quad sit there for 2mins USB-unplugged, it recovers and I can connect once again. If I switch MAG OFF, all problems are gone. So seems to do with MAG data handling/readut timing...

RipperDrone, 2016-06-27:
Confirmation: After switching MAG off in BFC 1.5, I can CLI dump out data and read back in data without any issues. Even read in older 2.8 dumps where I manually changed to mag_hardware parameter to NONE. As soon as I read in a dump that has mag_hardware = AUTO in it, the read sequence will complete, but the reboot will fail. 

Plugging USB out and waiting for 2mins then replug in allows for one reconnect where I can disable MAG again and cure the problem.

So it'S clearly related to MAG data readout / timings. Has not been any issue in all versions prior to 3.0.0.

RipperDrone, 2016-06-27:
Ok, what are your timings - mine are 2khz/2khz. Maybe if timing gets tighter, FC stmmbles over MAG readout / CPU load more severely...

gaelj, 2016-06-27:
I haven't ever managed to open the configurator so all the settings are default from clean flash.

RipperDrone, 2016-06-27:
@gaelj BF 2.7 / 2.8 were ok?

gaelj, 2016-06-27:
Didn't try those but 2.9 is OK.

AndersHoglund, 2016-06-27:
Good, you seem to habe narrowed it down to the Mag compass. My guessing now is that there is some config error in the new io system, causing i2c bus to be in a state where the mag does not have either sufficient power or speed to drive SDL or SCL  lines. Some open drain, pull-up or similar issue. 
Those that understand the new io system should have a look at the io pin configurations and compare to 2.9.0. I am lost.....

blckmn, 2016-06-27:
Does the spracingf3 have external pull up resistors? If not it's easy enough to enable the internal weak ones - I'll post code here shortly.

blckmn, 2016-06-27:
@AndersHoglund @gaelj @RipperDrone I just merged this for a test to the development branch. Can you pull it, compile and tell me if it works?

https://github.com/betaflight/betaflight/pull/625

blckmn, 2016-06-27:
https://www.dropbox.com/s/tjxhnf6qctexm7p/betaflight_3.0.0_SPRACINGF3.hex?dl=0

[18 more replies]

closed 2016-06-28 by blckmn