betaflight/betaflight #601: mid throttle sets all rc center values

salzamt, 2016-06-26:
hi,
I just set the middle throttle value in betaflight configurator 1.3 and 1.4 to 1440 as this is 50% throttle on my rc. this set the mid_rc to this value. due to that, all rc channels assumed 1440 as center. I dont know if this is the wanted behaviour but if so, middle throttle is kind of a misleading name for this option, took me a while to find that out.
feel free to just close this issue if this works as designed,
cheers, Riccardo

borisbstyle, 2016-06-26:
Why do you set mid throttle to 1440 actually? The name says itself......mid throttle is center throttle.

closed 2016-06-26 by borisbstyle

salzamt, 2016-06-26:
because it says "Middle Throttle [RC inputs center value]". thought for 3d I might give the throttle the real center value, which is 1440 on my RC. didnt know its affecting all channels.