betaflight/betaflight #607: Blackbox with OpenLog - lots of corrupted and empty Files on 2.9.0

SoulPictures, 2016-06-26:
Sense flashing mit Motolab to 2.9.0 my OpenLog crates a lot of empty and corrupted files, sometimes up to 30 within one pack.

Setup:
2 nearly Identical Quads

- Motolab Cyclone 8k/8k/8k, 
- OpenLog (SanDisk16GB Extreme & Transcend 600x)
- Aikon SEFM
- X4R (via SBUS & S-PORT)

Using 2.8.1 RC1 and 2.8.1 "Full" there was only one File per Boot including all flights.

Datalogging rate was set to 1/16 and 1/8, both show similar behaviour

borisbstyle, 2016-06-27:
Sounds like bad connection

SoulPictures, 2016-06-28:
You were right, sorry found a broken cable, has connection but loses connection as soon as vibrations occur....

closed 2016-06-28 by SoulPictures

adamgalmor, 2016-07-20:
I have the same issue but I'm using the on board cheap.
My setup is:
SPRacing F3 + Betaflight 2.9.0 (default settings)
Littlbee 20A BLHeli_S (firmware V16.1)
XSR (via SBUS & SmartPort)

P.S
With Betalight 2.7.1 the logging worked perfectly...

borisbstyle, 2016-07-20:
Lower the speed of logging. No point of creating issue about that. It is
blackbox related.

Just make the logging slower for openlog

gatmezeiz, 2016-08-17:
I experienced strange behaviour with 2.9 creating blackbox logs:
1. Quad: [F303](http://www.fpvmodel.com/x-racer-f303-flight-controller_g1106.html) FC, LB20, onboard dataflash and memory. At the first time I was able to configure it to internal datalash and save the created log, but next time when I tried to check the created log (after erasing, flying...) the configurator said that there is no onboard memory. I have lost the onboard memory!

2. Quad: TBS [Powercube](http://team-blacksheep.com/products/prod:powercube) with integrated FC (no onboard memory), configured to external logger (used to work before). At the first two times I got perfect logs but after the third time I was not able to read the SD card any more. I tried with another SD card, the first time it was ok, second time had the same issue, no card reader could recognise the card. So finally I have lost 2 SD cards :(.