betaflight/betaflight #608: beta 2.9......drone goes crazy!!!!!!!!

robyvalis, 2016-06-26:
I now installed on two drones beta 2.9, take off the drones go crazy and do somersaults 3000 ... much problem grve, the My two equal drones mount a fc racing f3, esc Littlebee pro, emax rs engines, with the beta 2.8. 1 did not have this problem.

blckmn, 2016-06-27:
have you a filter enabled?

borisbstyle, 2016-06-27:
Post your dump of settings alwas when you open an issue. Dont compare different drones. Downgrade to 2.8.1 and verify it its still issue. Sounds like you didnt assembly something properly

robyvalis, 2016-06-27:
Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# BetaFlight/SPRACINGF3 2.9.0 Jun 25 2016 / 23:43:54 (3d9b180)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO_RATES
feature RX_SERIAL
feature FAILSAFE
feature ONESHOT125

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB

# map
map TAER1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

[... 279 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 10
set rc_yaw_expo = 10
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 70
set tpa_rate = 20
set tpa_breakpoint = 1650

#

borisbstyle, 2016-06-27:
looks good. must be you connected something improperly. Check motor order
and prop direction

robyvalis, 2016-06-27:
Ora funziona...dopo essere ritornato prima a cleanflight 1.11.0 poi a 1.12.0 poi alla 1.13.... e poi direttamente a beta 2.9... e quasi funziona... poi muovo le impostazioni salvo...rimetto le impostazioni come erano...salvo...e  ...funziona.... la fc racing f3 di banggood ha problemi nei flash

KiteAnton, 2016-06-30:
@robyvalis I would suggest you post in english otherwise only a limited amount of people will understand you. 

Ran your comment through google translate and the issue seems fixed, closing.

closed 2016-06-30 by KiteAnton