betaflight/betaflight #610: I-Shaped F3 Controller from banggood

maxsymba, 2016-06-27:
Hi.

I bought two http://www.banggood.com/I-shaped-F3-10DF-Flight-Controller-5V-2A-BEC-Bent-
Straight-Pins-For-RC-Multirotors-p-1047005.html
In stock it flashed by modified BF SPRacingF3 2.3.4
But it can't flashed by clean BF SPRacingF3, nor by 2.3.4, nor by last versions.
FC can't start.
But... it seem that it can be flashed by BF SPRACING MINI firmwares.
Nevertheless there are some troubles with MINI firmware on this FC:
* the wrong mapping of motors (3,4 on 5,6)
* the inverted gyro axises: roll and pitch 
* problem with PPM-input
* no SPI
etc.

Can anybody navigate me - where I can fix  in code at least this problems?
Any other idea?
Maybe somebody have such FC and already worked with it?

Best regards from Ukraine, Max

blckmn, 2016-06-27:
You can try the ZCOREF3 target. It looks to be similar.

blckmn, 2016-06-28:
This controller almost certainly has the HMC5883 compass. There is an issue, and hopefully it is resolved by https://github.com/betaflight/betaflight/pull/626

blckmn, 2016-06-28:
I am closing this and it can be talked about in this issue: https://github.com/betaflight/betaflight/issues/597

closed 2016-06-28 by blckmn