betaflight/betaflight #612: betaflight 2.9.0, cannot disable oneshot 125 on Naze32

anakin323, 2016-06-27:
I tried to turn off oneshot 125 through GUI, and use command "feature -oneshot125" but nothing happened. Waiting for new update to fix this issue. Thank you.

borisbstyle, 2016-06-27:
there is no issue.

just use betaflight configurator and not cleanflight configurator ;)

blckmn, 2016-06-28:
set motor_pwm_protocol is the command (or the new GUI as @borisbstyle has indicated).

closed 2016-06-28 by blckmn

iamthestorm88, 2016-07-21:
hey guys! i just flashed over from cleanflight to the new Betaflight and after seeing how to disable oneshot125 in betaflight by utilizing the betaflight configurator i also installed that. my question if anyone can please help me is this. Im running a 210mm quad, with: 2205 Red Bottom Emaxx motors 2300kv, 30A lumeneer esc with active dampening and oneshot125, 5045 triblade props, PowerOSDPRO and some fpv gear and 4S battery 1300mah. my pids are stock. I tried to take off and it takes off and is actually damn near perfect besides this wicked ticking, clicking, jittering. I tried to lower my p values and all it does it make the quad fly sluggish. its really getting me down. a brand new quad and iv never experienced such a pain. i figured maybe power spikes or maybe oneshot was doing this, but i cant fix it. can anyone help me out? iv calibrated the ESCs and like i said set min and max throttle everything seems great except for this annoying twitching that the quad does at a hover or any time i move the throttle up or down. PLEASE HELP

jdphoneboy63401, 2016-10-03:
iamthestorm88, I had the same issue with the emaxx 2205 red bottoms... I had to go into blhelisuite, update and the go to the PWM output Dither setting and change it from default to 15. After that the chunking/jittering was much much better...

jdphoneboy63401, 2016-10-03:
iamthestorm88, I had the same issue with the emaxx 2205 red bottoms... I had to go into blhelisuite, update and then go to the PWM output Dither setting and change it from default to 15. After that the chunking/jittering was much much better...