betaflight/betaflight #624: Convert betaflight to use parameter groups for configuration storage

martinbudden, 2016-06-27:
See https://github.com/cleanflight/cleanflight/pull/2034 and related PRs.

mikeller, 2016-06-27:
Happy to start working on this one, how do you want to organise this, @martinbudden?

martinbudden, 2016-06-27:
@mikeller , generally it's not a good idea to have too many changes going on at once. I don't think there is any point starting on this until the F4 port is complete, that's why I've marked it as 3.1.

Also, @borisbstyle has not commented on this yet, so he may have other priorities.

mikeller, 2016-06-27:
Fair enough. Totally agree on 'not too many changes at once'. I wish there was a better roadmap for Betaflight, that would enable devs like me to pick up things high up on the to-do list.

blckmn, 2016-06-27:
I think we can start using issues, milestones and assignments to help out with that @mikeller

mikeller, 2016-06-27:
@blckmn:

> I think we can start using issues, milestones and assignments to help out with that @mikeller

Sounds like a plan.

I would like to contribute, but without any oversight it is difficult to identify useful chunks of work outside of the odd bugfix, and even then all work is running the risk of being duplicated by another dev for lack of knowing who does what.

borisbstyle, 2016-06-28:
Yes we should start using milestones more.

But realize that we changed a lot....I mean like really a lot. I will try to get 3.0 testing done soon. We need some public testing too.

martinbudden, 2016-06-29:
@borisbstyle , indeed - a huge amount has changed for 3.0.

borisbstyle, 2016-07-27:
@martinbudden
I will soon seperate master and dev branch again to continue working on this.
The master branch just needs few fixes for 3.0

martinbudden, 2016-07-27:
Yes, it will be good to see 3.0 released.

The big task I am working on at the moment is conversion of iNav to the new IO. When that is done I can start working on parameter groups. I don't think it should be too difficult to do the conversion to PGs.

borisbstyle, 2016-12-20:
I think this is pretty much almost finished and this issue can be closed

closed 2016-12-20 by borisbstyle