betaflight/betaflight #627: Failsafe don't stop Motors on Air Mode

TorbanDallas, 2016-06-28:
I've updated from BF 2.8 to 2.9, always used Failsafe Stage 2 Drop. Now on BF 2.9 the Motors don't stop spinning on Failsafe. I've tryed both Options for using the Air Mode (Air Mode always on and Air Mode on AUX Switch). Even on Stage 1 with Set Values for the AUX Channel the Motors don't stop spinning on Fail Safe.

borisbstyle, 2016-06-28:
when you disable feature airmode it should work right?

Make sure the switch is off....i mean switch should not go to airmode
position during failsafe.

borisbstyle, 2016-06-28:
btw did you wait long enough? Failsafe timers.

TorbanDallas, 2016-06-28:
On disabled Airmode it should work yes. 
Switch is off. I've tried the Failsafe by shutting down the Transmitter - like i always used to do bevor. 
I was waiting the Failsafe timer, right. On Failsafe the Motors turn down to the Airmode Speed but don't stop to spin. Even tried all possible Options and even got help by a friend - he didn't find any mistake on my settings. But he's still using BF 2.8., so he can't readjust that Failsafe Problem.

TorbanDallas, 2016-06-29:
i still don't know why Failsafe doesn't work on BF 2.9. Nothing on my settings or Hardware was changed.  I was trying around and found out, that BF doesn't notice when my receiver (Storm S603 Spektrum compatible) loses transmission. It looks like the receiver is sending the positions of my Transmitter sticks from the time at the binding process. So i've found a work around. I was turning down my Throttel way on my Transmitter so that BF would get just a rate of 925 points for the Throttel value, Binding my receiver again and after that changing the Settings on my transmitter back to 1000 points for the null Throttel position. So everytime i switch of my Transmitter now, BF get's the 925 points of Throttel (Like it was configurated at the Binding process). Than changed the failsafe minimum length to 935 points and now always the Failsafe works by switching of my transmitter.

now my friend updated to BF 2.9 too, using a frsky receiver with no problems.  

Well, i think that must be a special problem with Spektrum or Spektrum compatible Receivers or just the Storm receiver. Whatever, now it works with a work around.

KiteAnton, 2016-06-30:
Seems to be more of a tx/RX setting issue. If the receiver does not send something different to the fw when loosing connection to the tx then there is no way for the FW to know when to activate failsafe. If the receiver supports it then it is recommended to set it to "no pulses" at failsafe (for taranis this is done in the tx settings).

closed 2016-06-30 by KiteAnton