betaflight/betaflight #630: Motors jittering on low throttle

trojanGoat, 2016-06-28:
Here's the setup:
Betaflight 2.9 on spracing f3 @ 4khz
Multishot 14.6 on little bee 30a 
rs2205 / 2300kv

The motors are jittering on low throttle, even when min throttle is set to 1100 the jitters still happen.  If I throttle up even higher, the motors smooth out and it seems to fly fine. If the throttle drops while it's in the air, the quad will crash.  

This problem happens with airmode / no airmode, different motor timing settings, multishot, oneshot125.  When using the motor tab with the props off, I didn't notice the problem.  You can see in the video, the esc eventually went out of sync.

https://www.youtube.com/watch?v=CXu10dpcSWk

borisbstyle, 2016-06-28:
lower yaw P and/or yaw filter. I think your ESC's produce too much noise due to damped light. Try also for once without damped light

trojanGoat, 2016-06-29:
Hi Boris,

The lower yaw P and filter didn't really do much.  I tried lowering yaw to 40 and yaw filter down to 40 and tried raising to 200 as well.  Turning off damped light got rid of the jittering but who wants to fly without damped light :(   

In BLHeli raising demag compensation to high and adjusting motor timing to med-high seems to have made it worse.

I turned OFF pwm dither, that seems to have reduced the low throttle jittering substantially.  I also did some hover tests and it seems like the copter is flyable.  If I'm hovering in airmode and do rapid throttle up and down, the copter will do a quick twitch on 0-10% throttle (no death yaw)

trojanGoat, 2016-06-29:
Seems like this issue has already been posted... closing ticket  I will mount fc and motors on orings

https://github.com/betaflight/betaflight/issues/322

closed 2016-06-29 by trojanGoat