betaflight/betaflight #632: Yaw drift to right Naze32 rev's 5 and 6 Beta 2.9

pharaohrs, 2016-06-29:
I'm experiencing a yaw drift on beta 2.9 .

Same settings on 2.8.1 are OK .

Checked all center points , trims , rates ...

Is like the bug on nazes on 2.7 release fixed on 2.7.1 
Maybe like a gyro cal issue or gyro reset .

Tomorrow I will do more tests .

pharaohrs, 2016-06-29:
Drift corrected on trim by hour .

pharaohrs, 2016-06-30:
On Rev6 , luxfloat the yaw drift is more noticeable . i need six or seven points on taranis trim to correct it .

closed 2016-06-30 by pharaohrs

cbudd75, 2016-07-09:
I am having this same issue, just flashed to 2.9 last night and i have a yaw to the left on my SciSky board for my brushed micro

pharaohrs, 2016-07-09:
Try to ad yaw_deadband=6

Maybe this solve your problem

pharaohrs, 2016-07-10:
You do it ? Solved ? Let me know !