betaflight/betaflight #636: CC3D does not boot with 3.0.0

AndersHoglund, 2016-06-29:
CC3D does not boot with 3.0.0. 
All I get is 5 short and 2 long blinks, FAILURE_MISSING_ACC.
Using CC3D.hex, (non OPBL), and flashing via main port. Last test on development branch, Jenkins build 68 (2f10c77).

gaelj, 2016-06-29:
Possibly related to https://github.com/betaflight/betaflight/issues/597 which was solved in PR https://github.com/betaflight/betaflight/pull/626.
Does your build contain this PR ?

gaelj, 2016-06-29:
Actually no, that PR solved a MAG problem, yours is ACC related. But maybe the root cause is similar.

AndersHoglund, 2016-06-29:
Those I2C problems are probably unrelated, and yes #626 is on the dev branch since yesterday and included in later builds. I would suspect some similar IO-issue with SPI perhaps... Spent a few hours trying to see what the diff 2.9 -> 3-0 is. Nothing found yet though.

blckmn, 2016-06-30:
I'll have a look this weekend. Though I don't have a cc3d to test.

borisbstyle, 2016-06-30:
I broke my only CC3D a while ago :(

@gaelj 
Did you try both .bin and .hex?

gaelj, 2016-06-30:
Sorry I don't have any cc3d.

RuKeBo, 2016-06-30:
I am having the same issue with a Motolab Tornado.  I need to boot to DFU mode and reflash with 2.9.

AndersHoglund, 2016-06-30:
OK, MOTOLAB is also SPI_MPU6000.
Are there any FC with SPI gyro working OK ? 

Read somewhere that CC3D_OPBL also behaves the same. No real surprise though. Have not put op bootloader back and test that one myself.

blckmn, 2016-06-30:
I have a sort of SPRACINGF3 clone that is working fine, it has a 6500 connected via SPI. 

The x-racer is also SPI.

borisbstyle, 2016-06-30:
No tornado is mpu6050 actually.

Motolab cyclone works fine on 3.0 as I tested it last evening. Nothing
wrong with mpu6000

blckmn, 2016-07-01:
MPU6050 is i2c only...

blckmn, 2016-07-01:
CC3D also had the compass issue - can we confirm that its been tested on the latest build?

blckmn, 2016-07-01:
Also FuryF3 is a CC3D look alike no?

AndersHoglund, 2016-07-01:
My CC3D is "6DOF" or "Acro"-only. No Mag nor Baro. 
Even if I had a 10DOF it would not possible to test Mag since the board does not even start up, fails in acc/gyro detection.

I have tried checking all pin definitions, CS etc. And I can't see any diff from 2.9.0. But there is surely a diff somewhere.... probably a trivial thing once found.

nathantsoi, 2016-07-01:
If we add an f1 startup script that includes the hard fault debugger and
add the flag specified in the makefile this should make the error easy to
find. I can do this when I have a few min or if someone wants to try it
out, look at:

https://github.com/betaflight/betaflight/pull/374/commits/46a49f61470e4a34adc5998ae05fc966b4d65d41

borisbstyle, 2016-07-01:
what exactly isnt happening?

Is gyro not being detected? 2 flashes and than reboot?

AndersHoglund, 2016-07-01:
Exactly. Five quick/short and 2 long flashes. Continuously, it seems. I interpret that as FAILURE_MISSING_ACC.

RuKeBo, 2016-07-01:
I am not sure of the beep tones on mine (Tornado), but 5 quick and 2 long seems correct as I recall.  After that, my computer doesn't recognize that the board is there, does not show up in Device Manager, thinks it's some unknown USB device.  I tried everything to get it to recognize it but couldn't get it to work.  If I unplug it and plug it back in, the beep tones start again.

borisbstyle, 2016-07-01:
fix should be easy than. Probably some typo in target.h definitions for
gyro/acc definition I guess

I am not sure of the beep tones on mine (Tornado), but 5 quick and 2 long
seems correct as I recall. After that, my computer doesn't recognize that
the board is there, does not show up in Device Manager, thinks it's some
unknown USB device. I tried everything to get it to recognize it but
couldn't get it to work. If I unplug it and plug it back in, the beep tones
start again.

martinbudden, 2016-07-02:
Well, I've had a look and I can't see any obvious mistakes in the `target.c` or `target.h` files.

borisbstyle, 2016-07-03:
@martinbudden Me neither. This needs some debugging. Only those who have the target can do that. I dont have it anymore unfortunatelly.
Also COLIBRI_RACE target is broken. It goes into reboot cycle

AndersHoglund, 2016-07-04:
I have a J-link adapter on order, has not arrived yet. Trying to trace what is going on with LED-blinks (Doh...).
Come so far to see that  "mpu6000ReadRegister(MPU_RA_WHO_AM_I, 1, &in)" returns a value of "1", not the expected "MPU6000_WHO_AM_I_CONST (0x68)" in the "mpu6000SpiDetect(void)" function.
Time to bring out the scope and have a look at the SPI interface, unless someone has a clue why register reads fails over SPI. 
Initial clock divider has changed from 128 to 256, i.e. slower clock now. Tried both, no go.

borisbstyle, 2016-07-04:
Does the initial boot have to be done with slow SPI speed and sped up
later. The old code was doing that for CC3D

AndersHoglund, 2016-07-04:
Yes, initial slow boot was with divider 128 back then, now 256. Even slower. No idea why this has changed nor why it needs to be slow.

borisbstyle, 2016-07-04:
@AndersHoglund 
Other SPI targets suffer same problem btw. THey do recover after couple of cycles, but they have same issue

martinbudden, 2016-07-05:
@AndersHoglund 

can you try this change to see if it fixes the problem: PR https://github.com/betaflight/betaflight/pull/677

AndersHoglund, 2016-07-05:
Just tested this on CC3D. Sorry to say it did not help, still two long LED flashes. Too bad, I though you found it.

borisbstyle, 2016-07-05:
What are the SPI pins for CC3D? I cant see the defines so that means it uses default ones right?

Try adding them to the target file for example
```
#define SPI1_SCK_PIN       PA5     
#define SPI1_MISO_PIN         ?
#define SPI1_MOSI_PIN         ?

#define MPU6000_CS_PIN        PA4
```

martinbudden, 2016-07-05:
Default pins for SPI are defined in `bus_spi.c`, for SPI1 as used by CC3D MPU6000 they are:
```
#ifndef SPI1_SCK_PIN
#define SPI1_NSS_PIN    PA4
#define SPI1_SCK_PIN    PA5
#define SPI1_MISO_PIN   PA6
#define SPI1_MOSI_PIN   PA7
#endif
```

blckmn, 2016-07-05:
Its possible something else is clobbering PA5.

If it boots enought for a CLI - hit the command "resources" and see if anything comes up on PA5

borisbstyle, 2016-07-05:
@blckmn 
There is no command resources

Somehow this power cycling is not a good thing. I would rather prefer board to go into debug mode by being accessible through cli when these things happen and spit some kind of boot log

blckmn, 2016-07-05:
https://github.com/betaflight/betaflight/pull/683

@borisbstyle that's a good idea. will look into it.

martinbudden, 2016-07-05:
@blckmn ,

I managed to get it to boot to CLI by adding `#define USE_FAKE_GYRO` to the `target.h` file.
The command `resource` gave the following output:
```
# resource
IO:
A00:              PWM IN
A01:              PWM IN
A02:                FREE
A03:                FREE
A04:              SYSTEM
A05:              SYSTEM
A06:              SYSTEM
A07:              SYSTEM
A08:               MOTOR
A09:                FREE
A10:                FREE
A11:                FREE
A12:                FREE
A13:                FREE
A14:                FREE
A15:                FREE
B00:              PWM IN
B01:              PWM IN
B02:              SYSTEM
B03:              SYSTEM
B04:                FREE
B05:              PWM IN
B06:                FREE
B07:               MOTOR
B08:               MOTOR
B09:               MOTOR
B10:              SYSTEM
B11:              SYSTEM
B12:               FLASH
B13:              SYSTEM
B14:              SYSTEM
B15:              SYSTEM
C14:                FREE
```

borisbstyle, 2016-07-05:
@martinbudden 
Yes rousource command does work! I was trying resources LOL

Really nice! Naze32
```
# resource
IO:
A00:              PWM IN
A01:              PWM IN
A02:              PWM IN
A03:              PWM IN
A04:              SYSTEM
A05:              SYSTEM
A06:              PWM IN
A07:              PWM IN
A08:               MOTOR
A09:                FREE
A10:                FREE
A11:               MOTOR
A12:              BEEPER
A13:                FREE
A14:                FREE
A15:                FREE
B00:              PWM IN
B01:              PWM IN
B02:              SYSTEM
B03:              SYSTEM
B04:              SYSTEM
B05:                FREE
B06:               MOTOR
B07:               MOTOR
B08:               MOTOR
B09:               MOTOR
B10:              SYSTEM
B11:              SYSTEM
B12:               FLASH
B13:              SYSTEM
B14:              SYSTEM
B15:              SYSTEM
C13:              SYSTEM
C14:                FREE
C15:                FREE
```

blckmn, 2016-07-05:
I plan on cleaning up the resources command so it gives more information :) but its a work in progress.

borisbstyle, 2016-07-05:
@blckmn 
Through configurator would be awesome!

closed 2016-07-05 by borisbstyle

AndersHoglund, 2016-07-05:
Thanks for PR #683 , CC3D boots now. Closing....err Boris beat me to it.

RuKeBo, 2016-07-05:
Would this be included in build #365?  I just tried flashing that and still get same problem.

borisbstyle, 2016-07-05:
@RuKeBo 
No the jenkins is now out of sync with github due to my reorganisation of branches. betaflight needs to become master

bilson, 2016-07-05:
In this site I'm compiling latest master now: https://betaflight.qmd.cl/

RuKeBo, 2016-07-05:
Tried file dated 2016-07-05 18:52 and same thing.  Not sure if that is the newly compiled file or not.

AndersHoglund, 2016-07-05:
@RuKeBo . Yes PR #683 is included in build 365. You can allways match git commit id in the build artefact zip file name. In this case pr683 had commii id 60130cd and the build artifact is thus betaflight_3.0.0-365-60130cd.zip	
Normally you can also read the change log on the Jenkins site to see what is beeing built, but in this case when moving from betaflight to master branch there is no change log for 365.

RuKeBo, 2016-07-05:
Ok.  Well I tried build 365 and the file from the location @bilson recommended and both still won't boot.  I am using a Motolab Tornado though.  I also think I get 4 short beeps and two long, but it's hard to tell.

blckmn, 2016-07-06:
@RuKeBo that would appear to be a completely different issue - I have opened https://github.com/betaflight/betaflight/issues/686 for discussion on resolving it.