betaflight/betaflight #643: MAG not detected on SPRF3 EVO

gaelj, 2016-06-30:
This is not new to 3.0.0, 2.9.0 had it also. Haven't checked previous versions.
MAG is detected on CF and INAV on the same board.

Arakon, 2016-06-30:
It's disabled because it simply doesn't work right. There are a few issues with the Evo that would need fixing/implementation.

closed 2016-06-30 by gaelj

MJ666, 2016-06-30:
It was working on an AlienFlight F3 V2 prototype with an MPU9250 (SPI connection) after my last updates some days ago? Most of the exiting AF F3 V2 boards have an MPU6500 only. I would expect it to work on the Evo also when it will be activated in the firmware? At the moment the driver is even not build in to the Evo target. Ned's to be enabled in the taget.h file an recompiled. It's not yet working on the F4 targets. Likely some timing problems.

gaelj, 2016-06-30:
I see. Thanks for the info ! Hoping it will get merged soon.