betaflight/betaflight #647: Adjustments get deleted when you disable them

GuybrushX, 2016-07-01:
Hi,

I'm very new to tunable quads but I think it would be a nice feature if the Adjustments would be saved even when they get disabled.

For example I did set up some in flight adjustments on different channels for PID tuning. I wanted to disable them, do some testing and re-enable them -> all my adjustments were gone and I have to set them up again.

I'm using an Eachine Racer 250 with Betaflight 2.9.0 and a Tanaris configured to SBus (which is not flyable with default settings btw., I did change the motor layout already but that's another issue).

borisbstyle, 2016-07-02:
copy them from the cli so you can paste them back

GuybrushX, 2016-07-02:
Well that seems a bit complicated but works - if I know the right variables :-) I think I should back up the aux values, don't I?

May I ask you where your focus is with that project? Getting as much features as you want and/or having a nice GUI with most CLI commands easily accessible? I prefer a good GUI with tooltips and some hints on using them and with some really advanced features which are accessible from the CLI only. It makes it much easier for beginners like me to use the correct settings when the GUI gives hints, warnings, errors etc.

KiteAnton, 2016-07-02:
No aux is for the modes tab. adjrange is for adjustments.

closed 2016-07-03 by borisbstyle