betaflight/betaflight #650: SPR F3 2.9.0, Quad yaw and roll when cut throttle.

lhphuccc, 2016-07-02:
Hey guys, 

I went out to day with a new build. Right after i punch up and cut throttle, the quad yaws to the right and rolls left slight. I was on Float PIDC, quickly changed to Integer after that but still experience the same issue. I'll post a link in the comment below to the DVR footage.

Has anyone bumped into this before ? Please let me know how you solve this or what could be the issue.

Cheers and thanks in advance.

lhphuccc, 2016-07-02:
This is the cut of my DVR

https://youtu.be/06aZBoscWb8

KiteAnton, 2016-07-02:
Have you verified, at least twice, that you have airmode enabled?

lhphuccc, 2016-07-02:
Hello there,

I set my Airmode and Arm to 1 Aux switch and and i checked it a few times at the field too as i was changing the PIDs and PID Controller ... doesnt seem to help :((

erikmansson, 2016-07-02:
Can you provide a blackbox log?

lhphuccc, 2016-07-03:
oh man ... my FC have no Storage for logging, i'll figure out tmr when i head out to fly again and update you.

KiteAnton, 2016-07-06:
@lhphuccc any updates on this? Have you verified your TX/RX works properly and that airmode is active and the motors are running sufficiently? 
Haven't seen anybody else reporting this behavior.

lhphuccc, 2016-07-06:
Hey, i have been flying my quad just fine. Recently i upgraded to 2.9.0 and i ticked the "Airmode" always ON. Anyways, school was hectic and i couldn't manage to fly. Will report again ASAP. 

My guess is that my FC cannot keep up with the Gyro/PIDs rate. Is Float PIDC more demanding then the previous Rewrite/Integer ?

borisbstyle, 2016-07-09:
Closing this issue as it doesnt seem anything firmware related.
Discussions can always continue

closed 2016-07-09 by borisbstyle