betaflight/betaflight #662: Motors spun up full throttle on clicking Save

lavith, 2016-07-03:
After calibrating motor and ESCs to configure min throttle for 2.9.0. Right after clicking Save button in Betaflight configurator 1.5 all motors spun up full throttle for a sec and then shut off by itself. Very unwanted behavior.

Reverting back to 2.7.1 this phenomenon does not occur.

Software used : BetaFlight configurator 1.5 with firmware Betaflight 2.9.0 and target betaflight_2.9.0_SPRACINGF3.hex
Hardware used: DYS XMS30A flashed with 14.6 multishot, Motor: Xnova 2204 2300kv and FC is X Racer v2.1 from FPV model.

borisbstyle, 2016-07-03:
This is oneshot/multishot bug and not really a software bug. Always been
there.

Esc thinks that it gets a real signal during reboot of fc. When fc reboots
you never know what can be on the pins.

The manual clearly says REMOVE ALL PROPS when connecting lipo while
changing settings.
The best thing is to never connect a powered up system to configuration.

RipperDrone, 2016-07-03:
makes users facing pain & injuries if they're not prepared for it:

https://www.facebook.com/groups/FPVRacerGermany/permalink/711589945648572/?comment_id=711600035647563&reply_comment_id=711600942314139&notif_t=group_comment&notif_id=1467563465206732

Nothing to do about it? Outputs can't be forced to stay at LOW during FC reboot?

borisbstyle, 2016-07-03:
During reboot CPU doesnt do anything.....oneshot / multishot is based on
short pulses. Also not every esc / firmware will show this problem.

Just dont be lazy and remove props or use towel. It is explained and
mentioned in all manuals and even in the configurator. Safety first!

RipperDrone, 2016-07-03:
ok with me, was just thinking of newcomers...

reportedly BF2.8 with BLHeli 4.5 did NOT spin up motors whereas BF3.0.0 and BLHeli 4.6 does. 
If there'S nothing to do about it on the FC side (permanently grounding the outputs during boot - maybe  some specific pull-up / pull-down resistance setting?) then we can close this issue...

Maybe @mikeller or @blckmn can give a short comment - I memorize you've done a lot of FC hardcoding...

lavith, 2016-07-03:
I always have props off when I am trying to setup or test something. That being said, it troubles me that when I load betaflight 2.9.0 firmware, it happens. Load 2.7.1 it doesn't happen. Somewhere a long the line it triggers the bug in betaflight v2.9.0

martinbudden, 2016-07-03:
If I have space for it I like to power the flight controller off the balance plug and the ESCs from the XT60 connector. This means it is possible to power the FC without powering the motors. This is useful for some adjustments, but obviously not for calibrating the motors. (It also means you have to use a wiring harness rather than a PDB, but I prefer that anyway.)

marbalon, 2016-07-08:
It is easy fix for this just stop pwm on all motors and wait a while in reboot procedure. I don't have access to my sources now but this fix the problem in 100%.

lavith, 2016-07-08:
do i have to look at the source now?

borisbstyle, 2016-07-08:
@marbalon
There is already a wait of 1500ms I believe.

borisbstyle, 2016-07-08:
@marbalon 
Oh wait you want to do that prior to reset. Thats something else indeed. Did you do that in the driver library?

borisbstyle, 2016-07-08:
@marbalon 
see here
https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_msp.c#L1968-L1976

marbalon, 2016-07-08:
But in this solution there is no delay after stop motors, and ESC doesn't lost synchronization I think and like other people describe it is not working in last versions (maybe only in unsynced_pwm, I'm using this in all of my quads). 
But do this in systemReset function is saver because cover all scenarios. I have the same issue when I reset board from OSD (build in, not via MSP).

borisbstyle, 2016-07-08:
@marbalon 
Actually there is
https://github.com/betaflight/betaflight/blob/3d9b180b332f30a9d55e5fe7b8fd8d9327402bbc/src/main/config/config.c#L1062-L1063

borisbstyle, 2016-07-08:
And this is being triggered from configurator
https://github.com/betaflight/betaflight/blob/3d9b180b332f30a9d55e5fe7b8fd8d9327402bbc/src/main/io/serial_msp.c#L1925

marbalon, 2016-07-08:
Yes, this should works for reboot from MSP, but from LCD and build in OSD not, so maybe better fix is in reboot procedure ?

blckmn, 2016-07-09:
https://github.com/betaflight/betaflight/pull/704

should cover it off... but it will need some testing.

blckmn, 2016-07-18:
Can we please confirm this is all good now?

borisbstyle, 2016-07-18:
I couldnt reproduce this and I think i kept saving....flashing...messing
around with settings and rebooting like 200+ times with battery connected

blckmn, 2016-07-18:
ok - I'll close. if there is still an issue (which I can't reproduce now either) then we can look into it case by case.

closed 2016-07-18 by blckmn

marbalon, 2016-07-18:
Did you try do this with different PWM settings ? I have this issue every reboot using multishot + usnynced pwm 32khz, maybe it is related with this settings...

borisbstyle, 2016-07-18:
@marbalon
Did you test with 3.0 latest?

marbalon, 2016-07-19:
Will try.

si618, 2016-07-26:
Greetings, I also ran into this over the weekend. Here's a short video of the problem: https://youtu.be/EdjH9q3pAbc

Just tested a local build and confirmed it's fixed for me, thanks!

lavith, 2016-07-29:
I didn't do the test with latest 3.0 but went into BLHeli and turn Demag Compensation to OFF. This problem doesn't occur anymore with Betaflight 2.9.0

keryil, 2017-01-31:
I can confirm that simply switching to Unsynced PWM causes the motors to spin up during reboot for version 3.1.0.

Eganwp, 2017-01-31:
Please list Blheli firmware you have on the ESCs and what brand they are.

keryil, 2017-01-31:
Oh sorry about that, it is a Racerstar RS30Ax4 V2, a 30A 4-in-1 ESC. I tried on BLHeli_S 16 and 16.5, both versions gave the same behaviour. Also the speed of the spin up seems random every time a restart happens, sometimes only a couple will spin up slightly, whereas sometimes they will all spin up quite powerfully. 

That said, I burnt the ESC today for a completely different reason, so while I can't attempt any diagnostics now, I can check if it was just a faulty ESC or something when my new ESCs arrive. I'll make sure to drop a line either way.