betaflight/betaflight #664: F4 VCP issues

DaleKramer, 2016-07-03:
I have F4 REVO ACRO.

The last firmware on this F4 REVO ACRO was Raceflight BB235 which connected fine in RFC on a VCP com port that appeared when the FC was plugged in.

I have now flashed BF 3.0.0 onto my F4 REVO ACRO.

When I plug the F4 REVO ACRO into USB, device manger sees it and gives me TWO 'Other devices' called 'Revolution'. Both have no driver associated with them.

Because there are no COM port drivers for them, BFC does not connect, but the firmware is running because of the series of LED flashes I see when I plug it in.

In Zadig 2.2.689, they show up as Revolution (Interface 0) and Revolution (Interface 1).

If I assign (using Zadig) the 'USB Serial (CDC)' driver to EITHER ONE or BOTH (see attached image), I still can't connect in BFC.

I try Uninstalling both the drivers with 'Delete drivers ...' checked and the whole process starts over when I cycle the USB plug again (two Revolution 'Other devices' with no drivers)

I tried to reinstall of VCP drivers with this:
    Unplugged FC
    Ran CP210xVCPInstaller_x64.exe
    Plug in FC, still two driverless Revolutions, unplug FC
    Reboot
    Plug in FC, still two driverless Revolutions, unplug FC
    Ran a Repair session with VCP_V1.4.0_Setup.exe
    Plug in FC, still two driverless Revolutions, unplug FC
    Reboot
    Plug in FC, still two driverless Revolutions, unplug FC
    Ran a Remove session with VCP_V1.4.0_Setup.exe
    Ran VCP_V1.4.0_Setup.exe as a clean install
    Plug in FC, still two driverless Revolutions, unplug FC
    Reboot (in other words Restart the computer )
    Plug in FC, still two driverless Revolutions, unplug FC

Boris suggested that this was a known F4 VCP issue but I haven't found an issue about it, so here it is :)

borisbstyle, 2016-07-03:
I got mine working with Zadig! Had exact same issue.

You do have to select proper driver with zadig! You need the COM serial
one......forgot the exact name from the top of my head.

DaleKramer, 2016-07-03:
The only Zadig driver with Serial in it is 'USB Serial (CDC)' and as I said, I tried all combinations of that.  The image was when both Interface 0 and Interface 1 were using 'USB Serial (CDC)'

The VCP drivers are working since I can plug my iNav flashed SPracingF3EVO into it and it shows up as COM16.

borisbstyle, 2016-07-03:
@DaleKramer 
YEs CDC works for me!

borisbstyle, 2016-07-03:
And yes for me it also works on every F3 board with regular drivers

DaleKramer, 2016-07-03:
I am not doubting that, I hope you are not doubting me :)

blckmn, 2016-07-04:
Can you please try installing the VCP driver from STM:
https://www.dropbox.com/sh/y981obrxe2wwzo8/AACzMIHRFwyWLElxWDM2KwVka?dl=0

And let me know whether it worked?

DaleKramer, 2016-07-04:
1. Uninstalled the drivers I had installed by Zadig for both Revolutions (with 'Delete drivers' checked)
2. Unplugged FC
3. Ran the new dpinst_amd64.exe
4. Received message that the 'Revolutions' were not found.
5. Plugged in FC (two Revolutions without drivers were found.
6. Ran the new dpinst_amd64.exe, appeared to run successfully (no messages which seemed funny) and still both Revolutions remain driverless in device manager, even after cycling the USB cable on FC.
7. Looking further, I notice that when I downloaded stmcdc.inf, it had been renamed stmcdc.inf.txt
8. Renamed stmcdc.inf.txt to stmcdc.inf
9. Right clicked on stmcdc.inf and tried 'Install', not allowed
10. Ran dpinst_amd64.exe again, and the Revolutions disappeared and got message that driver was installed, replaced by a 'STMicroelectronic Virtual COM Port (COM20)' with yellow warning on icon (but now I KNOW we are almost there, yeah)  (not sure if FC was plugged in when I ran .exe, my bad)
11. Then things a little fuzzy as to what I did next, my bad.  But through a combination of unplug and run .exe, IT STARTED WORKING!!!!!

THANK-YOU @blckmn!!!

blckmn, 2016-07-04:
You actually only need the INF from Zadig and USBSER, but if that fails for whatever reason, the STM drivers also work a treat.

DaleKramer, 2016-07-04:
Thanks, case closed!

closed 2016-07-04 by DaleKramer

MiroRadenovic, 2018-01-09:
Thank you @blckmn !!!!!!!