betaflight/betaflight #666: COLIBRI_RACE target possibly broken

borisbstyle, 2016-07-03:
During target testing I discovered that my COLIBRI race board doesnt boot with 3.0 firmware. It does work on LUX_RACE firmware. THe hardware is same....colibri just has some extra stuff

martinbudden, 2016-07-04:
@borisbstyle , the COLIBRI_RACE and LUX_RACE `target.h` files are different. In particular the betaflight COLIBRI_RACE `target.h` file has references to the the 6000 gyro which are not in either the `LUX_RACE` `target.h` file or the Cleanflight COLIBRI_RACE `target.h` file. Perhaps these are cut and paste errors for when the target was converted to the new IO?

I'm not really familiar with the COLIBRI_RACE hardware, nor do I have a board, but I think it is worth you double checking the `target.h` file.

borisbstyle, 2016-07-04:
No I believe the guys from TBS were working on a new model with MPU6000.
They both have MPU6500 at the moment.

I will check again today and leave stuff away 1 by 1 in Colinri target to
see what is causing it.

borisbstyle, 2016-07-04:
Actually all VCP targets have issues I found out! My VCP targets get into a reboot cycle when first time connected and after 1 minute they seem fine. Only happens when connected to USB.

borisbstyle, 2016-07-04:
Yeah def VCP issue....LUX race and DOGE both have some issue. Actually it also sometimes happens without USB connected.......it keeps cycling for a while. So it seems some kind of init issue

readerror67, 2016-07-04:
Does it flash the ACC/GYRO failure LED code?
Had that happen to a board the other day and it somehow fixed itself.

borisbstyle, 2016-07-04:
@readerror67
Yes it always fixes itself when you leave it for a while. Once fixed you can stay connected forever and you can even save and reboot. It happens on cold starts

readerror67, 2016-07-04:
I actually had it hooked up to the debugger, the WHO_AM_I response was garbage, it was reading the same value each time it failed. Possibly something to do with SPI?

borisbstyle, 2016-07-04:
YEs it seems SPI issue! All i2c targets dont suffer this problem.

I think the SPI clock should be slow during init phase and than sped up after everything has been settled. That was the case on the old code

martinbudden, 2016-07-05:
@borisbstyle 

Can you try this PR https://github.com/betaflight/betaflight/pull/677 to see if it fixes the problem?

borisbstyle, 2016-07-05:
@blckmn 
Any idea why these SPI targets have issues with booting?

resource from COLIBRI_RACE. It takes like one minute to get out of boot cycle
```
# version
# BetaFlight/COLIBRI_RACE 3.0.0 Jul  3 2016 / 21:41:46 (7b1ea3f)
# resource
IO:
A00:               MOTOR
A01:               MOTOR
A02:               MOTOR
A03:               MOTOR
A04:              SYSTEM
A05:              SYSTEM
A06:                FREE
A07:                FREE
A08:              PWM IN
A09:                FREE
A10:                FREE
A11:                FREE
A12:                FREE
A13:                FREE
A14:                FREE
A15:                FREE
B00:                FREE
B01:                FREE
B02:                FREE
B03:              SYSTEM
B04:              SYSTEM
B05:              SYSTEM
B06:                FREE
B07:                FREE
B08:                FREE
B09:                FREE
B10:                FREE
B11:                FREE
B12:                FREE
B13:              BEEPER
B14:               MOTOR
B15:               MOTOR
C00:              SYSTEM
C01:                FREE
C02:                FREE
C03:                FREE
C04:                FREE
C05:                FREE
C06:               MOTOR
C07:               MOTOR
C08:               MOTOR
C09:               MOTOR
C10:                FREE
C11:                FREE
C12:                FREE
C13:              SYSTEM
C14:              SYSTEM
C15:              SYSTEM
D02:                FREE
F00:                FREE
F01:                FREE
F04:                FREE

```

borisbstyle, 2016-07-06:
This is how it boots. Takes about 40 seconds

https://youtu.be/O4IfUerxf_c

blckmn, 2016-07-06:
Its a long shot, and I can't test from my location at the moment, but if this works. https://github.com/betaflight/betaflight/pull/690 I'd be surprised, but happy :+1:

martinbudden, 2016-07-08:
@borisbstyle 

I think I have it this time. PR https://github.com/betaflight/betaflight/pull/703

[Explanation is at the bottom of PR 702]

martinbudden, 2016-07-08:
Fixed in PR https://github.com/betaflight/betaflight/pull/703.

closed 2016-07-08 by martinbudden

blckmn, 2016-07-09:
@martinbudden good find. I honestly don't know why i missed that - old age and i am going blind ;)

martinbudden, 2016-07-09:
@blckmn , actually I was thinking what a damn fine job you made of the conversions. You changed many many settings with virtually no mistakes.