betaflight/betaflight #670: Support buzzer in Kiss target

gregorg, 2016-07-04:
See #177 for the work done to port betaflight on the Kiss FC.

The kiss fc has a particular way to activate buzzer.

Felix from flyduino give me the following informations : 

> the uprt(throttle) is on TX2 pin, the PWM outputs are inverted (5v amp) its the same with the buzzer out (B13) the in is connected to the gate of a nchannel mosfet which is pulled up to 5V like that the buzzer should be connected between the buzzer GND pin and 5V. if the pinB13 goes high, the nfet starts conducting and the buzzer will beep :)

[image]

blckmn, 2016-07-04:
Can you try the firmware here?
https://www.dropbox.com/sh/1ednhcad9s3reop/AAC56aDNx7ZzvgL1r-a0-PLma?dl=0

I have inverted the BEEPER.

llambkin, 2016-07-04:
Let us know how you go @gregorg. I may just reimplement a buzzer back in my craft......

borisbstyle, 2016-07-04:
Also gyro INT pin is missing. It should be PB2. It now works unsynced

gregorg, 2016-07-04:
I don't know why, I can't get it work anymore .... whatever version I flashed, the blue led is blinking forever and I can't connect. Any clue ?

gregorg, 2016-07-04:
my bad, it was a driver issue with Linux ...

@blckmn I have uploaded your .hex file, buzzer is working. I mean, I can here buzzer beeping continuously, may be synchronized with the blink of the blue led.

However I can't test buzzer on a switch, I'm unable to communicate with a D4R-II in CPPM. The exactly same configuration works on Kiss fw RC27G, so it is not hardware related. Do someone test CPPM ?

llambkin, 2016-07-05:
@borisbstyle, does this mean that I can now run a different Motor sync to Gyro? Or what does it mean now that the INT is now assigned? Just curious

llambkin, 2016-07-05:
@gregorg, I will open another issue re. CPPM.

martinbudden, 2016-07-06:
@gregorg , is this fixed now? Can this issue be closed?

llambkin, 2016-07-06:
If @gregorg doesn't reply, I will test in about an hour and let you know.

gregorg, 2016-07-06:
As I said, buzzer is beeping as soon as I connect 5v, but I was not able to test with a switch on the radio because CPPM was not implemented.

If I test latest build can I verify that buzzer AND cppm is working ?

martinbudden, 2016-07-06:
@gregorg , yes, can you test this in the latest build. Remember boris has moved CPPM to the PITCH pin.

llambkin, 2016-07-06:
I just moved my RX to the pitch pad, and it doesnt work for me....

llambkin, 2016-07-06:
Also my Buzzer doesnt work at all. I have the GND of the buzzer on the buzz gnd pad, and the + on the 5v....

llambkin, 2016-07-06:
Build 85 BTW.... downloaded from here http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Development/

borisbstyle, 2016-07-06:
build 85 is not the right one. There was a fix later!

andmiller, 2016-07-06:
I'm on 85 and buzzer is not working for me either. It is connected as stated in the Kiss manual.

borisbstyle, 2016-07-06:
try this one
https://dl.dropboxusercontent.com/u/31537757/betaflight_3.0.0_KISSFC.hex

2016-07-06 15:51 GMT+02:00 Boris B. <dreambb1982@gmail.com>:

llambkin, 2016-07-06:
Still Nada on the PPM....

borisbstyle, 2016-07-06:
Pitch is PA2 pin right?

borisbstyle, 2016-07-06:
maybe KISS uses inverted PPM like anything else?

llambkin, 2016-07-06:
Well, It doesnt beep out to PA2. AUX beeps out to PA13 like the diagram above.....

borisbstyle, 2016-07-06:
No I was talking about PPM

martinbudden, 2016-07-06:
@borisbstyle , why don't you try putting it back on the shared pin and use the pwm_mapping trick to avoid timer conficts?

borisbstyle, 2016-07-06:
@martinbudden My main issue with KISS target is that we are doing this without any debugging. Just trial and error with users who are trying it out.
I would personally say to give KISS FC limited support for now. Kind of inefficient working like this.

borisbstyle, 2016-07-06:
@martinbudden 
Still buzzer should work without PPM trick

martinbudden, 2016-07-06:
@borisbstyle , I agree, trial and error is extremely inefficient. However, I thought we were almost there. Do you want to just try the shared pin, and then if that doesn't work we'll mark the KISS issues for v3.1?

I thought someone had got the buzzer working, maybe I misread a message.

llambkin, 2016-07-06:
Sorry, If I had a board to donate I would. I meant it doesnt beep out on a multimeter on PA2. AUX1 does beep out on the multimeter to pin PA13 on the CPU.

borisbstyle, 2016-07-06:
@martinbudden
Thats what I also read that it worked!

gregorg, 2016-07-06:
Hey guys, the build from @blckmn was OK, the buzzer was beeping !

borisbstyle, 2016-07-06:
Isnt the PPM timer thingy to prevent PWM and Oneshot125 timer clashing?

2016-07-06 16:33 GMT+02:00 Boris B. <dreambb1982@gmail.com>:

llambkin, 2016-07-06:
I confirm also, the @blckmn build is buzzing. but all the time, with the light output.

llambkin, 2016-07-06:
OK.....on further testing, it works on the @blckmn build. I plugged a battery in and it quit bitching.

borisbstyle, 2016-07-06:
Ok so buzzer works anyway. PPM doesnt work?

llambkin, 2016-07-06:
and I can assign a switch to buzz, and it works. So it was obviously the low battery alarm going off. Probably because the USB is being picked up as a battery source.

deepimpactio, 2016-07-06:
Is there some of this firmware for my wife?

llambkin, 2016-07-06:
Nice to see we can have a laugh in here :D.

llambkin, 2016-07-06:
@borisbstyle still nada on the PPM on the Pitch input.

martinbudden, 2016-07-06:
OK, I'm going to mark this issue as closed - the buzzer works.

Let's continue any discussion about CPPM in the CPPM issue. And if we don't manage to get that sorted today, I propose that it is deferred to 3.1. After all we want to get 3.0 out the door soon.

llambkin, 2016-07-06:
Do we know what fixed the Buzzer? the build that @borisbstyle didnt work to fix the buzzer.

martinbudden, 2016-07-06:
OK, which build fixed the buzzer, and where did you get it from?

gregorg, 2016-07-06:
this one https://www.dropbox.com/sh/1ednhcad9s3reop/AAC56aDNx7ZzvgL1r-a0-PLma?dl=0

martinbudden, 2016-07-06:
OK, so that is @blckmn 's build with an inverted beeper. We

borisbstyle, 2016-07-06:
I wasnt trying to fix the buzzer as I knew it was fixed in other PR,  but never noticed that it didnt get merged.
I am still puzzled about why PPM doesnt work.

martinbudden, 2016-07-09:
@blckmn , could you create a PR with your buzzer fix in it. Don't want the fix to get lost...

blckmn, 2016-07-09:
Hi @martinbudden its actually in the master branch, it was to simply invert the beeper output.

martinbudden, 2016-07-09:
@blckmn , thanks for clarifying that. It was unclear from the thread whether the fix had made it into the mainline.

I'll close this issue then.

closed 2016-07-09 by martinbudden

borisbstyle, 2016-07-16:
@blckmn 
I see beeper is inverted, but cant find the actual change done in the PR / commit history?

Kryptonit3, 2017-02-12:
@martinbudden @borisbstyle The beeper / buzzer function is inverted. Here is a demo video

https://www.youtube.com/watch?v=dN2dT0NUHxk

mikeller, 2017-02-13:
@Kryptonit3: I think there is either a problem with your wiring (have you hooked your buzzer up to 'BUZ GND' and '5V'?) or your board. I just did a quick test with a buzzer on a KISS FC, and it works all fine with 3.1.