betaflight/betaflight #673: KISS ESC Telemetry Support

llambkin, 2016-07-04:
Hello all,

I would like to know if KISS ESC telemetry can be supported.

Here is a PDF on how it works:
http://www.rcgroups.com/forums/showatt.php?attachmentid=8524039&d=1450424877

It may be able to be implemented on other FCs also.

llambkin, 2016-07-09:
I just thought I'd give you guys an update on where telemetry pins out on the board.

All 6 telemetry pads pin out to a diode that then goes to TX3 on the CPU. I think its a bit weird that it goes to a TX pin, since it is receiving data, not transmitting it. I guess that serial can send and receive down the same pin? and it mentions in the instructions that it must request the telemetry data before it receives it.

basdelfos, 2016-07-18:
The concept is pretty easy after all. There is a demonstration Arduino sketch which examples the concept for reading telemetry information from one ESC. The FW send a 30us PWM signal (via normal PWM wire) to the ESC and waits until it receives telemetry information back through the TLM signal wire. Reading the telemetry information from all ESCs can be done by calling ESC 1-4 in serial. 

1. Send 30us signal to ESC1
2. Wait until telemetry information is received from ESC1.
3. Process telemetry information of ESC1
4. Do step 1-3 for resp. ESC2, ESC3, ESC4.
5. Start over again.

Sounds not to difficult to implement in Betaflight.

It is strange that the TLM signal wire is connected to the TX3 pin, a RX should have been more logical. I guess the pin function assignment (in or output) can be changed by the FC FW.

Link to Arduino sketch: http://www.rcgroups.com/forums/showatt.php?attachmentid=8521072&d=1450345654

basdelfos, 2016-07-18:
Just checked with a multimeter; the TLM signal wire is connected to a diode/resistor, the other side of the diode/resistor is not connected to TX3 for sure (did not beep on multimeter). It looks like it is connected to the pins 1, 13, 32, 48, 64 (all of them are VDD(A)) of the STM32F303. VDD = Digital power supply, VDDA = Analog power supply. I guess thats just 3.3V, there need to be some other circuit from the diode to the FC that is transporting the serial signal. My hardware skills are not on par with software skills, will need to investigate some more, but don't know where to start. The traces on the board are not that easy to follow because of the tightness the board is designed.

fedorcomander, 2016-07-28:
@basdelfos you are so close :)))

IB1387, 2016-08-07:
Nice job guys, I wish you the very best luck with getting ESC telemetry to betaflight. 
The telemetry was the major reason for me to get a KISS FC, if you get the ESC telemetry working, I will be getting back on betaflight in an instant. I wish I could help more.
Thank you for your work!

basdelfos, 2016-08-10:
@fedorcomander can you tell which pin on the CPU is used for the ESC telemetry? Would be great, saves me a lot of time figuring out how the boards layout. Thanks.

fedorcomander, 2016-08-10:
I am legally not allowed to. sorry.

llambkin, 2016-08-10:
@basdelfos, did you want me to do an entire pinout of the board, then we can add it to the wiki or somewhere public to assist in development?

basdelfos, 2016-08-10:
@llambkin, that would be great and can also be used to add SmartPort telemetry to the AUX1 pad (see #955).

llambkin, 2016-08-10:
[image]
I forgot about this image that someone within KISS headquarters gave us

basdelfos, 2016-08-11:
Thanks @llambkin that will help, but still non of the labels references to the TLM/TRX port, so we still have to puzzel. Do you have a clue with pin the ESC telemetry is connected to?

I see that the UART3RX is missing in the image. According to the datasheet, the UART3RX should be assigned to PB11, which is labeled YAW in this image.

From what i have figured out while reverse engineering the board with a multimeter. The TLM pads are connected to a circuit equal to how UART3TX is connected.

basdelfos, 2016-08-11:
The ESC's sends telemetry information with 115200 baud, so this must be a hardware UART (SoftSerial is pretty much limited to 19200 baud).

llambkin, 2016-09-12:
Ok..... I really want to get this on the move, what do I need to find out about the board to make it easier to put the code together? I recently put the KISS board back in my craft, but I have to use KISS FW to get TLM data from the ESCs. I am still having issues with getting a good tune into the KISS firmware, and Betaflight seems to work fine even running stock PIDs.

I am pretty good at finding my way around a circuit board, but useless when it comes to coding. Just let me know what you want to know about the board, and I will go poking around. I have a scope now, so I can get more of an idea of what is coming out of where.

-Liam

llambkin, 2016-09-12:
I had a look around the board with a magnifier, and I can definitely report that it IS hooked up to the TX pad on the UART3 (via a diode, that I think is there to prevent damage to the STM if 5v is hooked up to the TLM pad), that pad is defined as PB10 in the code.

All TLM pads go through via's and essentially are all connected together, so I dont know how the KISS FW knows what ESC is providing what data....

IB1387, 2016-09-12:
The last question is easy I think:
The FC sends a pulse to one ESC, this then answeres to the FC with the telemetry data. The FC then proceeds to send a pulse to the next ESC and collects its data. 
Also I would absolutely LOVE to run Betaflight, but rely on ESC telemetry on my craft. I can't do much, but if there's something a Board owner can Do to help realize this feature, I am more than willing to help out.

fedorcomander, 2016-09-12:
Looks like you guys have all ingredients to cook some nice feature now ;)

llambkin, 2016-09-12:
Thanks @fedorcomander. I am really sorry that the KISS Firmware doesnt work out for me. I had real high hopes getting it working like many guys seem to have on YouTube. But Betaflight works pretty much right out of the box on this board.

I seriously dont know why I get the oscillations that I do on the stock firmware, and maybe once Felix actually releases the next stable release, it will be fixed.

IB1387, 2016-09-12:
The thing I don't really understand (just out of curiosity, I can't code for microcontrollers) is how the pulse is sent to the ESC. @basdelfos said it was sent over the normal PWM wire, but wouldn't that affect the throttle signal?
Hopefully, someone with actual C programming skills will take over and implement this. :)

llambkin, 2016-09-12:
@borisbstyle has a handle on it. As I get it, the pulse is 30us, which is smaller than the 42us of oneshot42, so essentially the ESC can see the pulses, and the PWM data independantly of one another.

borisbstyle, 2016-09-13:
@llambkin 
Yes this can be added in the future. Not just for KISS boards but also for other boards.

Basically 30us pulse to ESC, which will respond with 32bit packet with all data in it on TLM pad.

nathantsoi, 2016-09-15:
this looks really cool guys. it would be awesome to have this for other boards / escs. i read the spec, i think you're right @llambkin, we should just send the pulse then listen for the response. i dont get how it would work wired up to TX though?

[image]

at first glance, it seems like it could maybe be read w/ COMP5_INM comparator? (what does the `M` mean?)
http://www.st.com/content/ccc/resource/technical/document/application_note/4f/92/37/3e/55/66/46/e2/DM00074240.pdf/files/DM00074240.pdf/jcr:content/translations/en.DM00074240.pdf

idk enough about software serial to say if we can reuse that?

llambkin, 2016-09-15:
It is definitely pinned out correctly as per the board labelling too, I have used an OSD with tx to rx and rx to tx, and everything worked as it should. As for the above infos, I have no idea what it means. Maybe someone who has more of an idea can clarify....

beyersito, 2016-09-22:
Great that this feature is coming, im another kiss fc with betaflight willing to test, had to install it due to pads 1 and 2 not working, soldered to 3,4,5,6, custom motor mix and voila, also minim psd telemtry is working pefectly

basdelfos, 2016-09-30:
Off the record, started with coding last night. Seems not to difficult to code. Only thing is how i'm going to debug this...ideas? Can take a bit longer than i expect. The idea is to make a new task for ESC Telemetry.

And...have fried one of my KISS FC the last day...put the +/- wrong on my IRC VTX and everything is working (VTX, Cam, OSD) except the FC didn't survive it, it is totally dead 👎

nathantsoi, 2016-10-02:
on my mac, i usually use a st-link: https://goo.gl/K2P0rP and OpenOCD as described here: http://nathan.vertile.com/blog/2015/10/24/st-link-flash-and-debug-on-the-spracingf3-discovery/  here's how to wire it up: http://nathan.vertile.com/blog/2016/07/29/betaflight-kiss-flight-controller/#swd

basdelfos, 2016-10-02:
Thanks @nathantsoi. Which IDE do you use on your mac (i'm also a mac user)? I use Atom as it is a convenient editor with code highlight and some syntax checking. I use it for arduino programming.
I bought a st-link from AliExpress as BG is sold out.

basdelfos, 2016-10-02:
First try: https://github.com/basdelfos/betaflight/tree/esc_telemetry
The code runs on a Naze. Have to wait until the ST-Link and a new KissFC arrives.

If someone whats to code review it in the time being, be my guest.

To enable ESC Telemetry, type in the CLI:
`feature ESC_TELEMETRY`
`serial 1 1024 115200 57600 0 115200`

beyersito, 2016-10-05:
I have a custom motor mix, where motors one and two are on pads 5 and 6, could this work?

FPVZaphod, 2016-10-07:
Hi Guys, 
using the built-in current sensors of KISS ESCs by telemetry would save the weight of a current sensor - that's great. Maybe it would be useful to add another parameter: Static Load in mA. If this load, generated by video tx, receiver and other onboard stuff, is permanently added to the measured ESC current, it would give a pretty much accurate result for the OSD battery status. 
Thanks for your effort, great work!

llambkin, 2016-11-06:
Hello @basdelfos, are you able to rebase your build on the current master? I would like to test your code with D-shot

basdelfos, 2016-11-06:
@llambkin Done. You can pull this PR: https://github.com/betaflight/betaflight/pull/1457

It was a quick rebase, it builds but have not tested it yet. Have to go off to a testflight for an upcoming FPV event. Impression: https://youtu.be/lwk1S_ILotk

martinbudden, 2016-12-01:
Closing, implementing in separate PR.

closed 2016-12-01 by martinbudden

banderberg, 2017-09-18:
@basdelfos What is the BF 3.2 CLI command to enable KISS FC ESC telemetry without having to solder bridge TX3 and RX3?

mikeller, 2017-09-18:
`esc_halfduplex` I believe.

basdelfos, 2017-09-19:
@banderberg I always used the solder bridge on my KISS FC's (they are retired now and replaced for BFOSD capable FC's). The cli command is `set esc_sensor_halfduplex = ON`.
Please let us know if you get it to work.

banderberg, 2017-09-19:
@basdelfos That worked, thanks!

basdelfos, 2017-09-19:
Good to know. I will update the wiki with this information.

factoidforrest, 2025-02-01:
Hey sorry for bumping the dead thread, but is Kiss's telemetry protocol documented anywhere? I need to parse the sentences. Understand that theyre 115200 serial so thats a start

KissUltra, 2025-02-01:
TelemetrieBytes[0] = (Temperature&0xFF);
	TelemetrieBytes[1] = Voltage >>8;
	TelemetrieBytes[2] = (Voltage &0xFF);
	TelemetrieBytes[3] = Current>>8;
	TelemetrieBytes[4] = (Current&0xFF);
	TelemetrieBytes[5] = used_mAH>>8;
	TelemetrieBytes[6] =  (used_mAH&0xFF);
	TelemetrieBytes[7] = (rpm_tmp >> 8) & 0xFF;
	TelemetrieBytes[8] = (rpm_tmp & 0xFF);
	TelemetrieBytes[9] = get_crc8(TelemetrieBytes, 9);

uint8_t  update_crc8(uint8_t crc, uint8_t crc_seed){
	uint8_t crc_u, i;
	crc_u = crc;
	crc_u ^= crc_seed;
	for ( i=0; i<8; i++) crc_u = ( crc_u & 0x80 ) ? 0x7 ^ ( crc_u << 1 ) : ( crc_u << 1 );
	return (crc_u);
}

uint8_t  get_crc8(uint8_t *Buf, uint8_t BufLen){
	uint8_t crc = 0, i;
	for( i=0; i<BufLen; i++) crc = update_crc8(Buf[i], crc);
	return (crc);
}