betaflight/betaflight #674: SPRacing F3 Mini death roll on hard yaw

trojanGoat, 2016-07-04:
Hey Folks,

The problem does not happen every time I way hard, but after about a minute of trying to invoke the problem it will happen.  It looks like when I yaw left or right, one of the motors loses sync and the quad will go on a spirally roll or yaw of death.  This problem will also happen if I perform aggressive flips and rolls on zero-low throttle.  Is it possible that it's caused by using stick arming/disarming?

I've been having problems with this for some time now.  I've pretty much tried everything I can think of including other issues that's been noted on github.   

The setup:
spracing f3 mini - betaflight 2.9, soft mounted
lb 30a - blheli 14.6, no motors revered using software
rs2205 2300kv

Stuff I tried in Betaflight:
lowering yaw p gain 
lowering yaw filter
increasing min throttle
calibrate esc multiple times
anti_esc_desync 0,10000, 5000, 2500, 1250 (At 1250 the quad is getting floaty)

Stuff I tried in Blheli:
lower min startup power
pwm output dither =  off, 3, 7
motor timing = med, medhigh

Here's my dump info and link to blackbox:

{\rtf1\ansi\ansicpg1252\cocoartf1404\cocoasubrtf470
{\fonttbl\f0\fmodern\fcharset0 Courier;}
{\colortbl;\red255\green255\blue255;\red255\green255\blue255;\red0\green0\blue0;}
\margl1440\margr1440\vieww10800\viewh8400\viewkind0
\deftab720
\pard\pardeftab720\sl280\partightenfactor0

\f0\fs24 \cf2 \cb3 \expnd0\expndtw0\kerning0
\outl0\strokewidth0 \strokec2 \
# version\
# BetaFlight/SPRACINGF3MINI 2.9.0 Jun 25 2016 / 23:44:45 (3d9b180)\
# dump master\
\
# mixer\
mixer QUADX\
mmix reset\
smix reset\
\
\
# feature\
feature -RX_PPM\
feature -VBAT\
feature -INFLIGHT_ACC_CAL\
feature -RX_SERIAL\
feature -MOTOR_STOP\
feature -SERVO_TILT\
feature -SOFTSERIAL\
feature -GPS\
feature -FAILSAFE\
feature -SONAR\
feature -TELEMETRY\
feature -CURRENT_METER\
feature -3D\
feature -RX_PARALLEL_PWM\
feature -RX_MSP\
feature -RSSI_ADC\
feature -LED_STRIP\
feature -DISPLAY\
feature -ONESHOT125\
feature -BLACKBOX\
feature -CHANNEL_FORWARDING\
feature -TRANSPONDER\
feature -AIRMODE\
feature -SUPEREXPO_RATES\
feature RX_PPM\
[... 321 lines cut ...]
set p_level = 50\
set i_level = 50\
set d_level = 100\
set p_vel = 55\
set i_vel = 55\
set d_vel = 75\
rateprofile 0\
\
# rateprofile\
rateprofile 0\
\
set rc_rate = 200\
set rc_rate_yaw = 100\
set rc_expo = 0\
set rc_yaw_expo = 0\
set thr_mid = 50\
set thr_expo = 0\
set roll_rate = 65\
set pitch_rate = 65\
set yaw_rate = 75\
set tpa_rate = 25\
set tpa_breakpoint = 1750}

Here's blackbox log:
https://www.dropbox.com/s/f9ey5t96kj3fq7o/LOG00037.TXT?dl=0

borisbstyle, 2016-07-04:
Sorry but your issue is not fc neitgmher firmware. It is in the escs. Those
cause desyncs!  Sometimes raising min throttle helps

trojanGoat, 2016-07-04:
So both the little bee 30a esc and dys xm30a both have desync problems?  I thought it was just the dys esc.

deepimpactio, 2016-07-04:
I have seen many of my teammates roll back to BLHeli 14.4 and this goes
away

trojanGoat, 2016-07-04:
I tried that as well.  It didn't make a difference.

erikmansson, 2016-07-04:
Try setting demag to high in blheli

trojanGoat, 2016-07-05:
Alright.. I put up the min throttle and tested with 1 pack.. No death rolls.  I'll have to do some crazy flips and see if it holds up.

trojanGoat, 2016-07-05:
Just flew 5 packs around obstacles and I didn't get any death rolls.  I did use up a pack to try to invoke it one more time.  After about 2 minutes of smacking the yaw stick left and right, it caved and crashed.  I feel confident using the quad as a race machine but not so much for crazy acrobatics.  Lowering max throttle and increasing min throttle helped a lot.  ...closing ticket... Thanks Boris.

brousselle, 2016-07-06:
Well i discovered this entry yesterday, and i am facing the exact same issue as you with a very similar setup: 
SPRacing EVO, little bee 30a, Emax 2205 2300kv, air mode enabled.
I tried reverting blheli to 14.4, and also raised min_throttle up to 1100. If i do an agressive moves at min_throttle, the quad become crazy with yaw spiral.
Can you tell the values you have for min and max_throttle ?
Did you change demag value, or other stuff in blheli/betaflight ?

trojanGoat, 2016-07-06:
Actually I'm still having problems.  I tried demag low/high.  For a while it was barely doing the crashes and I thought that it actually all went away.  I noticed that when my props got worse, the deathrolls came back.  It only happens when I do fast rolls or fast yaw.  

my min throttle is 1060, max throttle 1955.   The only thing I haven't tried is reverting back to good old cleanflight.   Boris mentioned above that these esc's are known to cause desync issues.

brousselle, 2016-07-07:
So may be we should leave this issue open if you still have the issue. My machine is not flyable at all.
I will produce some blacklogs to see if can understand the root cause of this.
If not i will try to swap components: FC, then ESC to see where it comes from.

trojanGoat, 2016-07-07:
Ticket reopened.  I am still experiencing issues.  A few other things I'm trying...

1) lower min_check (so I can stick calibrate the gyro)
2) recalibrate gyro once the quad is on the floor and not moving at all.  @borisbstyle  Maybe I'm disturbing the calibration process when I attach my battery while the quad is upside down?  Does this make sense?
3) increase min throttle to 1075

I did another hover test and it didn't crash.  I'll have to do some flips and rolls to see if it happens again.

@brouselle if your copter is not flyable at all, you might be having a different issue.  Mine is totally fine and can even race with it as is.  The only time the death roll happens for me is when I'm doing some medium speed rolls and inverted rolls or very extreme and rapid yaw movements.  Maybe you should post a video with blackbox info.

brousselle, 2016-07-07:
Here is the test i did: (BF 2.7.1, BLHeli 14.6)
I changed min-throttle to 1150 in BLHeli and BF. but it does not change anything.
In my blackbox log, i do a rapid flip at 16s. Then motor 2 goes max_throttle, and seems blocked there until i disarm.
Visually (i do not have a video to show today) the quad is going crazy whereas we see in the log that the quad is only oscillating.
Is this a problem with my flight controller/gyro ?

borisbstyle, 2016-07-07:
No its your esc

brousselle, 2016-07-08:
Thank you Boris :D going to replace it and report back here.

minsu523, 2016-07-09:
Hi Boris, first of all, thanks for your work.
I have also dead roll issue with naze/f3(beta 2.7.1)+ dys20a/dys xm30a(14.6). I tried to solve this with all mentioned thing by user,  but I cannot. Still, dead roll is occured rarely...
Do you think it is specific issue only in dys esc??
What is your esc??
I feel last solution is change esc except dys.

borisbstyle, 2016-07-09:
It is in 99.9999% of the cases the esc when this happens.
Hard manouver desyncs are really typically what happens than.

borisbstyle, 2016-07-09:
Going to close this as it is not a firmware related issue. 
Discussions always can continue of course.

closed 2016-07-09 by borisbstyle

trojanGoat, 2016-07-09:
I do agree with Boris,  this is an esc issue.  I actually contacted FVT and told them about this problem.  They got back to me and told me to go back to **blheli 14.3**  I ran 2 packs hovering with the same stick motions that caused problems before and I had zero desyncs.  I'll need to go outside and do some crazy pirouettes and see if it happens again.

brousselle, 2016-07-14:
Just to give an update, i replaced my faulty ESC, but the issue is still the same.
I even have some motors that start or do not start randomly now.
I am going to redo the build with another Flight Controller (DOGE FC). I feel like there is something really weird with that SPRacing EVO.

trojanGoat, 2016-07-14:
If you think it's an FC issue, which might be the same for me, maybe you can open a ticket in three clean flight thread.  Dominic (clean flight guy) might have a solution.

brousselle, 2016-08-11:
to give an update here: I replaced and redo my build with another FC: replacing the SPRacing with DOGE FC.
Now it is working fine. Difficult to say at this point if i did something wrong in my previous build, or if there is something wrong with my SPRacing EVO. 
Most important: It is working now :D

NovysFPV, 2016-09-20:
Hi i have a same issue. You can see on my vid on ytb. Problem can be seen at 0:33 and 1:08. Little shake, my ESC is AIKON 20A and SPracing F3, Emax 2205 2300kv. I have tried a lot of thing as changing timming of esc, changing pids, changing min a max throttle, tried a lot of version of BF but didnt help. When i am not flying agresive without quick flips, quad flown really well. https://www.youtube.com/watch?v=0aXpAkMiLGw

trojanGoat, 2016-09-20:
Hey guys,

I started looking at my black box.  On top of all the vibration issues,  I realized that my d and p gain were both too high.  I lowered them down by about 10% and my copter flew a lot better.  Also, all my desync and death roll problems went away.

Just a quick summary:

1) soft mount the fc
2) use blheli 14.3 in lb30a
3) filter the gyro or lower p/d gain based on black box data

Hardware tips:
1) dal props are very heavy and become imbalanced after a crash.  I switched to 2 blade and I no longer experience desync.
2) rs2205 are noisy motors, they vibrate a lot after a few crashes

NovysFPV, 2016-09-21:
Hi dudes, today i finally solve my problem with ESC Desync. Just flash latest blheli 16.3 from my old 16.0. And quad fly like GOD!