betaflight/betaflight #675: BF 3.0.0 PWM motor control issue on CC3D Revo F4

RipperDrone, 2016-07-04:
On a CC3D Revo, motors do not respond in PWM mode, and it seems all channels are mixed up visa RacefFlight and BF 2.7 where all control channels are working fine and motors spin up nicely in PWM mode.

Has there been a major redesign in PWM controls and channels / pinout?

martinbudden, 2016-07-08:
@RipperDrone , can you clarify, CC3D Revo was not supported int BF 2.7.

RipperDrone, 2016-07-08:
Correct, no support in 2.7, but tried 3.0.0 dev version. PWM did not work, and ‎SBUS was confused. BF 2.7 tested on CC3D Standard (!) worked fine on PWM and default channel mix. Therefore question was why BF3.0.0 did not inherit PWM routines for the Revo from ‎2.7 CC3D Std...                                                                                                                                                                                                                                                                                                                                        Grüße / best regards, Reinhard                                                                                                                                                                                                                Von: Martin BuddenGesendet: Freitag, 8. Juli 2016 20:52An: betaflight/betaflightAntwort an: betaflight/betaflightCc: Reinhard; MentionBetreff: Re: [betaflight/betaflight] BF 3.0.0 PWM motor control issue on CC3D Revo F4 (#675)@RipperDrone , can you clarify, CC3D Revo was not supported int BF 2.7.

martinbudden, 2016-07-08:
@RipperDrone , CC3D standard and REVO are completely different flight controllers - REVO does not inherit from CC3D.

Can you describe the problem in a bit more detail?

blckmn, 2016-07-09:
Might need to look at the counting in the pwm_rx.c, 

```
    if (ppmCountShift > 1) {
    	ppmDev.currentTime = ppmDev.currentTime / ppmCountShift;
    }
```

There is no adjustment for shared timers, if the timer frequency is quicker (e.g. oneshot is active on a shared timer).

borisbstyle, 2016-07-09:
So there is a timer clash probably?

Is this still an issue?

martinbudden, 2016-07-09:
@borisbstyle , still an issue I think.

blckmn, 2016-07-09:
Definitely an issue. But only affects ppm / pwm... So it's ok if using serial. I know how to fix it, just waiting on a Revo to arrive to test.

RipperDrone, 2016-07-10:
I don't own a Revo myself, but a friend tried it:Changed over from Open Pilot, used PWM and found that motors did not spin up, receiver / FC were not assigned usual channels for controls. All rewiring (FC to receiver) and mmix attempts did not fix this.So we wondered whether or not PWM mode was properly supported on Revo yet.                                                                                                                                                                                                    Grüße / best regards, Reinhard                                                                                                                                                                                                                Von: Martin BuddenGesendet: Samstag, 9. Juli 2016 00:35An: betaflight/betaflightAntwort an: betaflight/betaflightCc: Reinhard; MentionBetreff: Re: [betaflight/betaflight] BF 3.0.0 PWM motor control issue on CC3D Revo F4 (#675)@RipperDrone , CC3D standard and REVO are completely different flight controllers - REVO does not inherit from CC3D.

Can you describe the problem in a bit more detail?

borisbstyle, 2016-07-10:
This will be fixed soon. It indeed seems that f4 pwm timers are not correct

Tomcatt, 2016-07-13:
Still a no-go on that last pull.  Motors won't spin.

borisbstyle, 2016-07-13:
Can you try the very latest....few min ago

Tomcatt, 2016-07-14:
Yeah, bro anytime. I'm going to put my old IM MS 14.6 back on and test that too.

Tomcatt, 2016-07-14:
Okay, I’m not sure but it’s like there’s a disconnect somewhere with the software talking to the motors.  I’ve tried different combinations to get them to spin.  All I can do is get them to tweak and that’s it.  I see my arming lights on the computer and on the board and that’s it.  No-go.

I wish I’d knew something of coding to give you guys a hand.

The ESC’s are booting up and ready to go and that’s all.

Tomcatt, 2016-07-15:
3.0 still has the same issue with motors not spinning.  Now the ESC's are singing at different times.

llambkin, 2016-07-16:
This is most definitely still an issue. I have just changed over from a KISS FC on my quad, and I am having the same result. My ESCs are singing the start tone, but its like as if there is no PWM output. I have KISS ESC's and they light up, then go out...

llambkin, 2016-07-16:
Also @borisbstyle it might be also related to the timers, I have no idea, but the VBAT isnt showing correctly. When I plug USB in, it reads correct, but when I plug a battery in via a divider, it shows 25v. I have checked the voltage on the divider, and it reads around 2.5v, so its definitely within the 3.3v max.

llambkin, 2016-07-16:
OK, I have sorted my VBAT issue. It might be my divider. I had to set the scale all the way down to 56 to get it accurate.

Tomcatt, 2016-07-16:
Okay my CC3D ATOM EVO; I had to flash it with the .hex file it.  I also now own a CC3D that’s the step below the REVO. Both ATOM and normal CC3D will spin with MS and UNSYNC turned off.  Oh it sounds like ass if you spin the motors with unsync PWM turned on.  4k and higher with unsync PWM and the ESC’s won’t see the FC.  I’ll finish testing with a transmitter at a later date.

The REVO still has no joy running the motors.  At first I thought I was going to have to go with way of the little CC3D’s because they worked better with unsync PWM turned off.  I’m still not sure on the why yet.

Why the motors sound bad on the lesser CC3D I think because they are F1 boards.  They may not be able to handle the task at hand because of the max speed of 72MHz whereas F3/F4 runs at 168MHz.  I may have to play with that more.

blckmn, 2016-07-16:
@llambkin can you pull the latest and try. There is an issue with sync versus unsync if you are running 4khz or higher you would notice it. Try unsyncd now, with a frequency of 400 and calibrate, then update the frequency to 2000.

blckmn, 2016-07-16:
@Tomcatt F3 also run at 72mhz

llambkin, 2016-07-17:
@blckmn, I have tried running at 1khz, I knew there was an issue with 2khz and above. I will try it again and let you know.

I will say that I couldnt even get any beeps when calibrating on 2khz or higher, but with 1khz I can get a beep at the top but not at the bottom.

Just an observation. When I pulled the buffers off my KISS FC, I had the same result until I removed the inversion on the PWM output. The F4 isnt outputting inverted is it?

llambkin, 2016-07-17:
Ok, I have tried the latest code this morning. All the ESCs light up and make the right tones, but still no action. When I move the motor test slider up and down it get confirmation beeps from the ESCs but no spinning.

I am running 1khz and have tried oneshot125, 42 and pwm. None work.

blckmn, 2016-07-17:
Just tested mine at 1, 2 and 4khz use pwm_sync on and off for all, and motors all spin up and down no issues.

blckmn, 2016-07-17:
do you have a scope?

llambkin, 2016-07-17:
No, sorry I dont. I will do some more in depth testing soon.

llambkin, 2016-07-17:
I have done some in depth testing. I flashed Raceflight and that works as it should. I had to set the looptime to 2000 to calibrate the ESCs, they make the right beeps and spin the motors when calibrated.

When I flash back to BF, and set the looptime to 1khz, and leave the motors as synced, I get:

Correct startup tone and lights on ESCs lit.
On motor test page, I move the slider to the top and the light goes out, move it to the bottom, and it comes back. But no movement from the motors.
If I try do a calibration, the lights are out completely until I move the slider down.

When I up the looptime to 2000 (synced) I get:

No lights on the ESCs when the battery is plugged in.
Nothing works on the motor test page.
When I try calibration, the ESCs go into calibration mode, but when the output is lowered, it doesnt acknowledge the lower point.

When I leave the looptime at 1khz and unsync the output to 400hz, I get:

Nothing at all, the light stays out, and they dont make the acknowledge tone for either calibration or seeing a PWM signal.

Finally, When I set the output as PWM, and unsync to 400hz, I get:

Motors will spin, but the output on the motor slider is back to front. top is bottom, and bottom is top. It actually calibrates back to front also.

I hope this helps
-Liam

llambkin, 2016-07-17:
And yes I am running the current code. I did a sync just this morning.

borisbstyle, 2016-07-17:
Perhaps there are timer clashes on Revo target.

At least this is how it sounds.

blckmn, 2016-07-17:
@llambkin what are you using for your RX? Serial, ppm or pwm.

llambkin, 2016-07-17:
Serial sbus.

RipperDrone, 2016-07-17:
Just to complement the current state of discussion: When we tested on a Revo it was in PWM mode, not any seria mode. Quad didn't arm. Raceflight running fine, though.

borisbstyle, 2016-07-17:
@RipperDrone
Can you retest with the latest code. A lot has been changed and fixed since your latest tests

blckmn, 2016-07-17:
I'll double check the target, but given its serial there should be no timing issues.

RipperDrone, 2016-07-17:
I sent the last code to my mate Sascha who has the Revo quad. Will test PWM soon.

blckmn, 2016-07-17:
The fixes today won't solve timer clashes. We will need to solve those separately. If using serial they shouldn't be an issue.

Tomcatt, 2016-07-17:
I didn't have any issues with S.Bus on the REVO.   You released a new build? Thanks guys.

RipperDrone, 2016-07-17:
Tried today's sources build on the CC3D REVO in PWM mode - not yet there...

- all channels of RC were correctly assigned this time, including AUX channel (only 1 used/tested). All ranges show 1,000-2,000
- still all motors on EMAX Simon Series ESCs (BLHeli BL) did not start, not even sound any startup tune after power cycling the FC. 

Flashing back to RaceFlight - no problem. Running PWM on LibrePilot: No problem. 
In BetaFlight REVO target, there still seems to be something wrong with the init sequence and/or timings?

borisbstyle, 2016-07-17:
@RipperDrone 
Serial mode would work I expect. PPM / PWM modes need workarounds for timer sharing. Not sure if it applies to REVO, but I know a lot of other boards require that.

RipperDrone, 2016-07-17:
@borisbstyle 
thanks, we will try. Fried a motor an hour ago when trying to flash latest BLHeli onto the EMAX Simon Series 12A ESCs in order to prepare rewiring of the FC and then run serial mode. Used an ATMEL USB linker, but the EMAX have Atmel chipsets - motor started twitching then went up in smoke. So it'll take some repair time before we can shift to serial mode ;-(

Anyway, pls let us know when the PWM/PPM fix will be implemented - or are you planning to deprecate PWM modes in general?

llambkin, 2016-07-17:
@bluecheese, are you saying that your board flies and works? If so, what
board are you running? Mine is an "acro" board. It has no compass, baro or
RF module. I don't know if that makes a difference.

llambkin, 2016-07-18:
Well..... I have no idea what I just done..... But I got some spinning happening.....

Can someone please tell me what I did to get the motors working. Please find attached picture.
[image]

llambkin, 2016-07-18:
Well. Celebration short lived...... yes the motors spin, but they dont start reacting to sticks except throttle until the throttle reaches 1500, then it reacts to the gyro and other stick movements. 

Kinda screwy..... But oh well, I guess I will wait until someone here chimes in......

borisbstyle, 2016-07-18:
There is a new option....zero stabilisation thingy. Put it to ON for now

borisbstyle, 2016-07-18:
But yeah sounds like some timing clashes there

llambkin, 2016-07-18:
Well.....That go it :D
Wish me luck for the hover test.....

llambkin, 2016-07-18:
OK! AWESOME! You can now call me Captain Hackjob! I have no idea what I am doing, but it now hovers :D

llambkin, 2016-07-18:
@borisbstyle, just a quick question..... Is what I have done dangerous? Can it just flip out whilst in flight with no chance of recovery? Maybe the fact that I have to ask this question tells the story?

-Liam

blckmn, 2016-07-18:
We may have missed a target.c update somewhere along the lines - good spotting.

blckmn, 2016-07-18:
@llambkin can you pull the latest, it should be fixed.

llambkin, 2016-07-18:
Thanks :D
I will do a sync, and see if it turns out the same as mine.

llambkin, 2016-07-18:
Yup, all good. Hovers here in the garage. I will flight test in the morning, and report back.

RipperDrone, 2016-07-18:
@llambkin: Have you tested serial, PPM or PWM mode to be hovering ok (in a garage ;-))?

blckmn, 2016-07-18:
I'm pretty sure this is now resolved (motor control).

RipperDrone, 2016-07-18:
We'll try out asap - still licking our wounds from last crashes :-)

llambkin, 2016-07-20:
Definitely resolved @blckmn I have successfully put 3 packs through this board now.....and not a glitch. running 8k 8k 8k.

llambkin, 2016-07-20:
I am also using all 3 serial ports. 1x for blackbox, 1x RX serial, 1x OSD.

borisbstyle, 2016-07-20:
Still need to figure out why KISS motor 2 and 4 dont like the synced pwm

blckmn, 2016-07-21:
@borisbstyle theres another issue for Kiss - I'll close this one and we can continue to look at Kiss in that issue. I think I need to order one.

closed 2016-07-21 by blckmn

Tomcatt, 2016-07-22:
I’m looking at your work on Jenkins and would like to know how to download an older build?  I want to test out build #356.

Thanks