betaflight/betaflight #676: Support CPPM for KISS FC Target

llambkin, 2016-07-05:
See #177 for the work done to port betaflight on the Kiss FC.

On the KISS FW, you use the same pins to connect an RX to. Weather this be CPPM, Serial RX etc. Can this be mirrored in Betaflight? Or will people have to connect to another input eg. Roll or Pitch?

@gregorg and myself would like to know. As per the diagram below, the Pitch input is PA2 T2C3.
https://cloud.githubusercontent.com/assets/541602/16553214/53f0a2a0-41c7-11e6-804e-4ca25fad330e.png

martinbudden, 2016-07-05:
@llambkin , I think it is desirable to keep the same pin usage in betaflight and the KISS FW - that way people can change between betaflight and KISS FW without rewiring. So I'll look into the CPPM pin issue. Also, does KISS FW have softserial? If so, on which pin(s)?

borisbstyle, 2016-07-05:
@martinbudden
I didnt add PPM as it was really meant to test the target first,  which
works now so now the rest can be completed.

KISS FC doesnt have softserial. It is also not able to use esc passthrough
due to buffers

martinbudden, 2016-07-05:
@borisbstyle , can you add the PPM? I'm not really that familiar with the KISS hardware, so I might make some silly mistakes if I try and do it. Also softserial on AUX1 and AUX2

llambkin, 2016-07-05:
Softserial on aux1 and 2 sounds good. @borisbstyle, I don't know if it has been done on betaflight yet, but you would have to use serial rx2 for PPM.

Maybe KISS FW inadvertently had softserial, since they use aux1 for telemetry that would be awesome if this can also be achieved in BF. I miss my Amber whingeing about my battery level.

I currently run OSD, blackbox and serial RX. So telemetry got the axe....

borisbstyle, 2016-07-05:
@llambkin @martinbudden KISS FC uses the same pin for Serial RX and PPM. So when PPM mode activated serial becomes PPM. That is not really common for cleanflight/betaflight targets and surely not the way how current code base works. That's why it is not added. I think the majority uses serial nowadays anyway.

Unfortunatelly I broke my KISS Fc a while back so I cant do much troubleshooting on the hardware. For telemetry you dont need softserial. You can also just use the other UARTs

borisbstyle, 2016-07-05:
I think it is easier to add PPM on another pin. The KISS layout for soldering makes that flexible.

llambkin, 2016-07-05:
Stick it on the Pitch pad, its right next door, and can be done easily.

borisbstyle, 2016-07-05:
Ok but traditional PWM doesnt work anyway. KISS target will only support PPM and serial than for now

Actually in the current code the pin definition is correct, but it conflicts with serial

borisbstyle, 2016-07-05:
Try this one
https://github.com/betaflight/betaflight/commit/b274289f0c8233be35dbf857026cf3a4409937b7

MJ666, 2016-07-05:
Sparky and AlienFlight F3 also share the same pin for Serial RX and PPM. Implementation is not that complicated. For having compatibility of used pins for different Firmware options it would be good to not alter the pinout in BetaFlight if not needed. Specially if users solder the cables directly and don't use plugs.

borisbstyle, 2016-07-05:
@MJ666 
Actually I didnt even noticed this
https://github.com/betaflight/betaflight/blob/betaflight/src/main/drivers/pwm_mapping.c#L307-L312

I guess KISS can be added to do the same, but we get more target specific stuff in the code. Should find a neater way to bring this into target definitions

gregorg, 2016-07-05:
I have just tried b274289 with PPM soldered to PITCH pin, it doesn't work.

borisbstyle, 2016-07-05:
You are sure that the MPU INT change was working?

borisbstyle, 2016-07-05:
This is the change I mean
https://github.com/betaflight/betaflight/commit/c7c459d122fd4f55cae47680ddba1dee3dbce2ac

gregorg, 2016-07-05:
No I'm not, how to validate it ?

borisbstyle, 2016-07-05:
@gregorg 
Download that particular build 
http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Development/

Its the build 82 you need to verify and if that one doesnt work the one before that

borisbstyle, 2016-07-05:
@gregorg 
Oh wait.....you were actually saying the board works, but just PPM doesnt work? sorry for misinterpratation

borisbstyle, 2016-07-05:
aghhh I see.....I know why it doesnt work. Let me change that

gregorg, 2016-07-05:
yes, PPM doesn't work and buzzer doesn't bip anymore, I'll wait for your next commits :)

martinbudden, 2016-07-05:
@borisbstyle , on this particular issue, I wouldn't be worried about adding target specific code into `pwm_mapping.c`.

`pwm_mapping.c` is already choc full of target specific code, adding one more item won't make any difference. For 3.1 I'll try and look at a way of cleaning that whole area up.

borisbstyle, 2016-07-05:
@gregorg Try again the latest betaflight build

@martinbudden 
I dont want to mess a lot now. I dont have the target to debug and test. At least there is now PPM :D

borisbstyle, 2016-07-06:
maybe KISS uses inverted PPM like anything else?

llambkin, 2016-07-08:
I think it might have something to do with the fact that I cant use a multimeter to beep it out....Id bet that if we assigned PPM to AUX1 (next to pitch pad), that it would work fine. God knows where Pitch pins out to. I cant seem to find the pad it goes to.

borisbstyle, 2016-07-08:
@llambkin 
Pin was properly repurposed in the code. If you cant beep it than something else is in between.

llambkin, 2016-07-08:
I was thinking that. I will have to go inverter chasing, Id bet there might be a signal inverter there somewhere. Sorry, I'm not in the shed tonight, or I'd do it right now.

llambkin, 2016-07-09:
I got round to it this arvo. It goes through a 10kohm resistor on the underside of the board. I dont know how or weather this affects weather we can use it for PPM, maybe it attenuates it to a point where the CPU cant read it. We might be forced to either use the RX2 where serial RX is, or use aux 1 or 2.

I just beeped the ROLL input on the underside of the board, and that is directly connected to the CPU. It could be used, but it would be an inconvienience, since you have to unbolt your FC to solder to it. The ideal situation would be to have it on the serial pads so everyone doesnt have to resolder.

I have no idea why they used a resistor for the PITCH input only.

llambkin, 2016-07-11:
I wanted to make some changes myself, so I downloaded the entire workspace, and I cant do a make TARGET=KISSFC on it....... It complains about LT0 version expected 4.1, I have 3.0 apparently. I dont understand, so I guess I will have to wait until someone gets around to it.....

I just wanted to put PPM on AUX1 so I can get my little Vole 165 up in the air.

borisbstyle, 2016-07-16:
@llambkin 
Can you try the latest code?

Assigned PPM to AUX2
https://github.com/betaflight/betaflight/commit/50da1cfa1defba9bc08520a3a03ec05c2efdec2f

llambkin, 2016-07-17:
> 50da1cf

Will do, it would be good to get my little 165 back om BF :-D

llambkin, 2016-07-17:
Sorry, I have been really busy lately. I will get around to it this week. Unless someone else gets to it before me......oh hang on, my vole is packed REAL tight, I cant get to AUX2 without completely disassembling the quad. AUX1 would be a better option if possible?

Just search up Vole165 and see why..... its SUPER tight in there.....

llambkin, 2016-08-10:
Can this be closed now? PPM appears to be working now. And also RX serial not affected.

borisbstyle, 2016-08-10:
Yeah i finally managed to get a kiss fc and flew it today on 3.0. Works great

closed 2016-08-10 by borisbstyle