betaflight/betaflight #687: Fury F4 Frsky Telemetry not working

Rossbow, 2016-07-06:
I can not get VFAS telemetry working on the Fury F4

borisbstyle, 2016-07-09:
Is this still an issue on the latest build?

blckmn, 2016-07-09:
@kc10kevin any ideas?

Rossbow, 2016-07-09:
@borisbstyle  I just tried the latest build and still not working.
I connect the single wire from the x4rsb to uart3 tx, then set smart port on uart3 in ports tab and enable telemetry in features, This is the same way I set up all my F3 boards

blckmn, 2016-07-09:
@Rossbow smart port requires either soft serial, or an inverter, F3 processors have a built in inverter. F4 don't.

Rossbow, 2016-07-09:
Ah OK, Well that sucks

blckmn, 2016-07-09:
https://blck.mn/2016/06/smartport-the-frsky-x4rs-and-betaflight/  gives a hack if you are good with an iron.

closed 2016-07-09 by blckmn

Rossbow, 2016-07-11:
@blckmn  Thank you, I did the hack and all worked perfect

jrwrodgers, 2016-08-28:
@Rossbow Can I ask you some questions about how you got this hack working? Which UART did you use? Was it 1 wire or 2 wires? Any chance of a picture? I cant get it to work yet.

Rossbow, 2016-08-28:
@jrwrodgers  I am using a FuryF4 FC and  I used UART3 and I used 2 wires,  1 to RX and the other to TX.
As for pictures, I just followed this guide [https://blck.mn/2016/06/smartport-the-frsky-x4rs-and-betaflight/](url)

jrwrodgers, 2016-08-28:
Many thanks for the response...I have just got it working following this method in the last hour! :)