betaflight/betaflight #694: Vbat in fight error

flavien317, 2016-07-06:
Hello,
I have  SP F3 FC with the last Betaflight (2.9.0) and I use vbat on my quadcopter. When I pug the batterie and the USB to Betaflight, it shows me the right value, but if I fly and put throttle, it's decreasing very fast (I check with a battery voltage). In my OSD I always have "Low Voltage" alarms on,even after 30s - 1min of fly.

I'm the only one ?

llambkin, 2016-07-07:
Yeah, its called voltage sag. It doesnt matter what FC or batteries you are running. I have a brand new 65C Dinogy, after about 30 secs, if I do a punchout (5-10sec), it will tell me the battery is low. After the punchout it recovers fine, and continues to have plenty of power for a good few mins. The best way to monitor batteries is to use a current sensor, and only fly to 80% of the battery. And keep away from sustained full throttle punches. This kills lipos.

I blamed my battery problem on the fact I changed from betaflight to KISS FC, I was addament that it was the FC and their problematic FW. It was only then, I did a test with my Fluke meter, hoping to prove it wasnt my batteries (they werent that old) only to find, yup, all of my batteries sag.
https://www.youtube.com/watch?v=rg-U1h6M5TY

flavien317, 2016-07-07:
Ok, thanks for the reply. So what I need to have the correct vbat even if I fly ?

llambkin, 2016-07-07:
That depends on what you call correct. You are likely receiving what the
actual voltage of your battery is. On MWOSD there is a filter you can add
to get a more smooth battery reading. Just uncomment FILTER_HYSTERYSIS and
// the standard filter.

martinbudden, 2016-07-08:
Closing, not a bug.

closed 2016-07-08 by martinbudden