betaflight/betaflight #696: STM32F4 / SmartPORT doesn't work

Mistral, 2016-07-07:
Hey,

I have own FC with STM32F4, it is flashed by BF 3.0.0, onboard I have inverter. I have seen inverted signal from XSR receiver on UART1_RX pin (PB7), but nothing happens on UART1_TX pin (PB6). I have in target files defined "TELEMETRY" and 

`#define USE_USART1
#define USART1_RX_PIN PB7
#define USART1_TX_PIN PB6
#define TELEMETRY`

In BFC is selected SmartPort with AUTO on USART1

Why don't it work propely?

blckmn, 2016-07-08:
Smart port requires both a Tx and rx pin.

Mistral, 2016-07-08:
Yeah, I know it. It is done by 2 channel inverter with opendrain outputs with diode on TX. On RX pin I see inverted signal, but TX doesn't send anything.

blckmn, 2016-07-09:
I have smart port working great. I'm not sure there's a Betaflight issue here.

blckmn, 2016-07-09:
Perhaps post some wiring diagram or config you have.

borisbstyle, 2016-07-10:
I have smartport working great too on different targets.
Surely no issue there

llambkin, 2016-07-21:
My REVO doesnt output S.Port on USART3 either.... I have wired it  using both TX, RX and TX+RX.
Also tried with and without telemetry_inversion = on and off

nikybiasion, 2016-07-23:
Try to set pid loop to half of gyro update and see if it works

blckmn, 2016-07-26:
@llambkin and everyone else. Remember the F4 doesn't have inversion built in to the MCU UART. You need to either hack the Rx unit (to remove the inversion), use an inverter cable, or use softserial.

This is the same as the days of F1 and Sbus inversion issues - hopefully FrSky will see reason an release a Rx that has a non-inverted smart port option.

I'm going to close this issue as Smartport is working fine with the above use cases.

closed 2016-07-26 by blckmn

Tomcatt, 2016-08-12:
@borisbstyle Bro, what did you do to get it working?  Trying to get the Sparky 2 setup for this.  On CF page says just tie it into the TX port (main or flex I'm sure).