betaflight/betaflight #697: NAZE32 data flash size 0 reported on 3.0

mikeller, 2016-07-07:
With my NZAE32, the size of the onboard data flash is reported as 0 (as seen by the configurator in the response to `MSP_DATAFLASH_SUMMARY`:

```
> DATAFLASH
Object {ready: true, supported: true, sectors: 0, totalSize: 0, usedSize: 0}
```

For some reason the data flash is recognised, is reported as ready, but returns a size of 0.
﻿

AndersHoglund, 2016-07-07:
I have soldered a 16MB flash on my Naze32 rev5 Acro, works fine on 2.9 but fails to be detected on 3.0.0

    # version
    # BetaFlight/NAZE 2.9.0 Jun 25 2016 / 23:43:09 (3d9b180)
    # flash_info
    Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216, usedSize=0

    # version
    # BetaFlight/NAZE 3.0.0 Jul  6 2016 / 22:32:07 (7fab6b9)
    # flash_info
    Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=256, totalSize=0, usedSize=0

mikeller, 2016-07-07:
@AndersHoglund: Sounds like an interesting mod. Does the chip go onto the empty spot on the 'acro' version board? Also, do you have the spec and source of the chip you've used?

AndersHoglund, 2016-07-07:
I got a 10 pack of Winbond W25Q128FVSG from AliExpress a year ago, can't recall exactly what dealer. Similar to those: http://www.aliexpress.com/item/Free-Shipping-10pcs-W25Q128FVSG-25Q128FVSG-W25Q128-SOP8-IC-FLASH-128MBIT-104MHZ/32636840651.html?spm=2114.01010208.3.188.ftLkP6&ws_ab_test=searchweb201556_8,searchweb201602_2_10017_407_406,searchweb201603_8&btsid=50b2eb00-9b39-42b2-adfe-eb25c20a668e
The one I go was not exactly the correct package, a bit too wide to match the PCB footprint. Had to bend the legs underneath the chip. Not sure what package type the Naze32 flash footprint is made for...

AndersHoglund, 2016-07-08:
I got my Naze32 rev5 data Flash working by changing SPI CS pin definition to PullPull. Not sure this is the correct way, even though also CC3D MPU and Falsh works fine with this patch.
In 2.9 SPI CS is defined as OD, and I can not find any special case for it.... 
The patch:

    $ git diff
    diff --git a/src/main/drivers/bus_spi.h b/src/main/drivers/bus_spi.h
    index 6404690..dbba948 100644
    --- a/src/main/drivers/bus_spi.h
    +++ b/src/main/drivers/bus_spi.h
    @@ -30,7 +30,7 @@
     #define SPI_IO_AF_SCK_CFG     IO_CONFIG(GPIO_Mode_AF_PP,       GPIO_Speed_50MHz)
     #define SPI_IO_AF_MOSI_CFG    IO_CONFIG(GPIO_Mode_AF_PP,       GPIO_Speed_50MHz)
     #define SPI_IO_AF_MISO_CFG    IO_CONFIG(GPIO_Mode_IN_FLOATING, GPIO_Speed_50MHz)
    -#define SPI_IO_CS_CFG         IO_CONFIG(GPIO_Mode_Out_OD,      GPIO_Speed_50MHz)
    +#define SPI_IO_CS_CFG         IO_CONFIG(GPIO_Mode_Out_PP,      GPIO_Speed_50MHz)
     #else
     #error "Unknown processor"
     #endif

AndersHoglund, 2016-07-08:
With the scope I can see that SPI2 CS line is driven high with 2.9.0, Not so with 3.0.0 
In bus_spi.c (2.9.0) there are a few special cases (for each SPI 1/2/3) that does this: 

    #ifdef SPI2_NSS_PIN
        // NSS as gpio slave select
        gpio.pin = SPI2_NSS_PIN;
        gpio.mode = Mode_Out_PP;
        gpioInit(SPI2_GPIO, &gpio);
    #endif

All this seems to be missing in 3.0.0.
 
Frankly, I don not really understand all of the conditionals in 2.9.0,  nor in 3.0.0 to fix this.  In 2.9.0 it seems to be if "SPIx_GPIO" is not defined then "SPIx_NSS_PIN" is defined and based on that SPIx_NSS is set to "Mode_Out_PP". I don't follow this "logic" chain.....

In 3.0.0 there is a similar chain based on the new "SPIx_SCK_PIN" -> "SPIx_NSS_PIN" . But there it stops, no further output mode settings based on this. 

Should we re-introduce all the old code as of above example ( and more) ? Someone cleaned it all away, maybe for a reason. Was the thought to move it somewhere else? As target specific piece of code?

borisbstyle, 2016-07-08:
@AndersHoglund 
Interresting findings

I was thinking to try commit by commit from beginning to find the moment where it got broken. But it certainly seems that all SPI based stuff is only having quirks.

martinbudden, 2016-07-08:
@AndersHoglund - very good spot. The change to the code was due to the conversion to the new IO, but this looks like an inadvertent change.

@borisbstyle "I was thinking to try commit by commit from beginning to find the moment where it got broken" - I think if you did that, you'd just find it was in the commit that converted to the new IO, which probably was quite a big commit.

I think what is required is reading through the code checking, line by line, that everything is set up correctly. Either by comparing with the old code, or comparing with a reference manual. [It's fairly clear that NSS needs to be set to output pp.]

I can spend some time over the weekend doing that.

AndersHoglund, 2016-07-08:
What if we invert the conditionals? I.e. My patch sets CS to "GPIO_Mode_Out_PP" unconditionally . What are the conditions when this should NOT be set? Any such case?? With F3 and F4 it is already unconditional this way...

Maybe the simple thing is just to set PP? I could make my patch a PR.....

martinbudden, 2016-07-08:
@AndersHoglund , the current code is:
```
#elif defined(STM32F1)
#define SPI_IO_AF_SCK_CFG     IO_CONFIG(GPIO_Mode_AF_PP,       GPIO_Speed_50MHz)
#define SPI_IO_AF_MOSI_CFG    IO_CONFIG(GPIO_Mode_AF_PP,       GPIO_Speed_50MHz)
#define SPI_IO_AF_MISO_CFG    IO_CONFIG(GPIO_Mode_IN_FLOATING, GPIO_Speed_50MHz)
#define SPI_IO_CS_CFG         IO_CONFIG(GPIO_Mode_Out_OD,      GPIO_Speed_50MHz)
```
the current Cleanflight code is:
```
#ifdef STM32F10X
    gpio_config_t gpio;

    // MOSI + SCK as output
    gpio.mode = Mode_AF_PP;
    gpio.pin = SPI1_MOSI_PIN | SPI1_SCK_PIN;
    gpio.speed = Speed_50MHz;
    gpioInit(SPI1_GPIO, &gpio);

    // MISO as input
    gpio.pin = SPI1_MISO_PIN;
    gpio.mode = Mode_IN_FLOATING;
    gpioInit(SPI1_GPIO, &gpio);

#ifdef SPI1_NSS_PIN
    // NSS as gpio slave select
    gpio.pin = SPI1_NSS_PIN;
    gpio.mode = Mode_Out_PP;
    gpioInit(SPI1_GPIO, &gpio);
#endif
```

So it looks like your proposed change is absolutely correct.

AndersHoglund, 2016-07-08:
Great, thanks.

closed 2016-07-08 by AndersHoglund

borisbstyle, 2016-07-08:
@martinbudden
Are the F3 gpio definitions also the same like in cf?

Still need to fix booting issue with boards connected on SPI

martinbudden, 2016-07-08:
@borisbstyle ,

I checked the F3 definitions and I couldn't find any differences. But I will check again - it's very easy to miss something.

martinbudden, 2016-07-08:
@borisbstyle , I think I've found it, see: PR https://github.com/betaflight/betaflight/pull/702

Unfortunately I don't have the facility to test the at the moment, so could you check it?

mikeller, 2016-07-08:
Yes, #702 fixed it, data flash now works for NAZE. Thanks @martinbudden.

martinbudden, 2016-07-08:
@mikeller , it was @AndersHoglund 's PR that fixed it.

mikeller, 2016-07-08:
Oh, my bad. Thanks @AndersHoglund in this case.

AndersHoglund, 2016-07-08:
OK, no probs. My pleasure. I was a bit confused first, how #702 could fix this...