betaflight/betaflight #700: SPF3 CPU reports 100% when acc_hardware set to NONE in CLI

dalgibbard, 2016-07-08:
Hey all,
Bit of a weird issue, though not hugely relevant perhaps.
Using Betaflight 2.9.0 on an SPF3, Multishot, unsynced PWM @ 32000, 4k/4k.

With acc setup as defaults; 28% CPU usage.
Setting ```set acc_hardware = none``` results in 100% CPU usage.
Disabling Acc via the Betaflight configuration Tab in the configurator, 14%.

Seems also that if you disable it with configurator, and then set to 'auto' with CLI, also results in 100%.

Why does the old CLI command break stuff? Should it?

```
# status
System Uptime: 3 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:100%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n
Cycle Time: 395, I2C Errors: 5, config size: 1296

# tasks
Task list:
0 - (SYSTEM) max: 2us, avg: 0us, rate: 9hz, total: 0ms
1 - (GYRO/PID) max: 277us, avg: 230us, rate: 2463hz, total: 4sec
2 - (ACCEL) max: 135us, avg: 85us, rate: 327hz, total: 181ms
3 - (ATTITUDE) max: 48us, avg: 17us, rate: 79hz, total: 17ms
4 - (RX) max: 107us, avg: 75us, rate: 45hz, total: 20ms
5 - (SERIAL) max: 626us, avg: 2us, rate: 80hz, total: 8ms
6 - (BATTERY) max: 11us, avg: 0us, rate: 39hz, total: 0ms
7 - (BEEPER) max: 8us, avg: 0us, rate: 76hz, total: 0ms
```

borisbstyle, 2016-07-08:
you need to re-power the board after changing that stuff sometimes

closed 2016-07-08 by borisbstyle

dalgibbard, 2016-07-08:
No worries; thanks for confirming Boris. Much appreciated!