betaflight/betaflight #709: Hybrid Dterm calculation method suggestion

etracer65, 2016-07-09:
@borisbstyle 

I was thinking about the benefits/detriments between the different Dterm calculation methods and thought that a hybrid approach might me useful.  The "error" method is great for sharp stick response (due to the "kick") but not so great at actually doing it's job in the PID controller since it doesn't respond directly to the gyro.  The "measurement" method is better for managing the actual PID response and does a better job around transitions (like the end of moves, prop wash, etc.) because it's seeing the actual gyro data, but it's "softer" because there's no coupling to the stick.

So how about a hybrid method that is primarily based on measurement, but utilizes the **derivative** of the stick movement to "accelerate" the Dterm?  The rate of change in the stick becomes the accelerating factor not the magnitude of the stick change.

borisbstyle, 2016-07-09:
I have this in the pipeline for testing anyway with some other things.

Still work in progress.

etracer65, 2016-07-09:
Thanks - can't wait to try it.

A few other thoughts on this:

I'm a little concerned about the negative acceleration when the stick is centered.  I'm wondering if the negative Dterm acceleration will momentarily mess up the Dterm's ability to cleanly stop moves.  After the negative kick the tracking would return to the gyro but I'm concerned it might cause an undershoot on the stop.  Maybe the acceleration should only be "positive" or increasing the current rotation.  Have to try it and see I guess.

Also using the derivative of the stick movement should eliminate most of the Dterm "kick" other than the initial acceleration (the desirable kick).  The steps in the rc_command shouldn't manifest as Dterm kicks.  Maybe this will help quiet down those clamoring for rc_smoothing to come back! ;)

borisbstyle, 2016-07-12:
I dont want to go through the same pain again like before with rc smoothing :)
Besides that refresh rate isnt really solid among the tx systems. Taranis tends to have different refresh rates on different channels depending on settings. Most go for huge channel configurations.

How do you mean you are afraid of the negative acceleration?

It will take a while before I really continue with this as I have some other potentially better methods to solve the kicks when stick centered.

etracer65, 2016-07-12:
Take the example of a quick full stick roll.  On the initial stick movement from center there will be a big derivative of the stick movement because it's accelerating and this will give the desired "acceleration" to the Dterm.  I'm concerned that when the stick is quickly returned to center to end the move the large "negative" derivative might have an undesirable impact on the Dterm stopping the roll effectively.  Since the measurement method does such a great job in this case (by getting the delta from the gyro) I'm not sure we want this extra influence from the stick return in this case.  Don't really know for sure without testing but it's worth thinking about.

borisbstyle, 2016-07-12:
Well negative delta is used in measurement and does a smoother job on
stops, but it does leave more error for iterm accumulation.

But there are also other ways to smooth out those stops and not just delta.
Testing and analysing is the key to determine the best options.

>
> Take the example of a quick full stick roll. On the initial stick
movement from center there will be a big derivative of the stick movement
because it's accelerating and this will give the desired "acceleration" to
the Dterm. I'm concerned that when the stick is quickly returned to center
to end the move the large "negative" derivative might have an undesirable
impact on the Dterm stopping the roll effectively. Since the measurement
method does such a great job in this case (by getting the delta from the
gyro) I'm not sure we want this extra influence from the stick return in
this case. Don't really know for sure without testing but it's worth
thinking about.

etracer65, 2016-07-12:
I'm up for testing! :)

Actually I wasn't referring to negative delta as that's certainly needed for the Dterm to do it's job.  I was referring to the big negative "acceleration kick" that would caused by the derivative of the stick moving quickly back to center (the reverse of the desired "kick" when the stick first increases in rate).  I'm thinking this might upset the Dterm calculation momentarily as it's working to get the smooth stop (which already works very well with measurement).  Since it's looking at the acceleration of the stick I imagine the effect would be short lived so maybe not an issue, but testing will tell.

borisbstyle, 2016-08-02:
done in 3.0

closed 2016-08-02 by borisbstyle

etracer65, 2016-08-04:
@borisbstyle 

Can you describe how this works and/or how to use it?  Using the latest 3.0RC2 and development configurator 1.71 (43929cb) I don't see anything that indicates a hybrid delta calc - just the previous measurement and error methods.  Also I recall you mentioning in the regroups forums about a "slider" that would allow adjusting how much error was applied (I think) but I don't see anything like that.

borisbstyle, 2016-08-04:
Configurator stuff is still work in progress and will be there in final
release.

But its there. dterm_setpoint_weight

0 = measurement
0.5 (50) = half error
1 (100) = error like before
2 (200) = double error accelerarion
3 (300) = triple error ;)

You get the picture. You can use any value in between

Try default values. It is already configured well to give you very nice
stick feel.
Value of 2 with smoothed rc is insanely nice on stick

etracer65, 2016-08-04:
Ok, thanks.  How does it interact with pid_delta_method?  Does that still determine whether it's measurement or error - or is it always measurement now and the amount of error influence is set by term_setpoint_weight?

If it still switches between measurement and error methods, what's the impact of term_setpoint_weight when the method is set to error?

I unfortunately won't get a chance to fly until the weekend, but looking forward to trying. :)

borisbstyle, 2016-08-04:
Pid delta method is only for legacy pid controller.
Doesnt do anything on the new pid controller.

etracer65, 2016-08-04:
Thanks - that makes sense.  I guess it will all become clearer with the configurator is updated.  Hopefully the delta method parameter will disappear when the new PID controller is selected to minimize confusion.

borisbstyle, 2016-08-04:
Exactly.

2 new parameters /sliders will appear there.