betaflight/betaflight #712: PWM Mapping cleanup

blckmn, 2016-07-09:
Place holder for Pwm mapping cleanup in 3.1

I'm thinking we ditch it entirely, and use motors, servos, pwm inputs etc and use ioRecs, timerRecs etc to define what pins do what. Then the resource cli command (and an associated msp) can simply allocate a function.

borisbstyle, 2016-07-10:
That would be excellent! lets start that with the CLI first.

savaga, 2016-07-10:
I've done something similar here - https://github.com/savaga/betaflight-sirinfpv/commit/aedc8712da023298c2d226808c495b1df37f4c23

blckmn, 2016-07-10:
Hi @savaga thanks. What I was looking for is essentially persisting the ioRecs and timerRecs (these would need to be added) into configuration, i.e. owner, resource and index (would need to be a unique combo).

The resource command can then set them. Upon initialisation instead of IOInit() setting these values, it (or another named method) effectively retrieves the IO_t (or TimerInit() returns TIM_t) item required.

Then in target.c or target.h you set the default initialisation for IORecs and TimerRecs for fix config or reset config methods.

If something already has an owner, then IOInit will return IO_NONE - effectively disabling the function (e.g. sonar, ws2811 etc).

blckmn, 2016-07-10:
As an example:
```
static IO_t  ws2811IO  = IO_NONE;
static TIM_t ws2811TIM = TIM_NONE;

void ws2811LedStripHardwareInit(void)
{
    ws2811IO = IOByFunction(OWNER_LED_STRIP, RESOURCE_OUTPUT, 0);
    IOConfigGPIO(ws2811IO, IO_CONFIG(GPIO_Speed_50MHz, GPIO_Mode_AF_PP));

    ws2811TIM = TIMERByFunction(OWNER_LED_STRIP, RESOURCE_OUTPUT, 0);
    TIMConfigOC(ws2811TIM, ....);

    TIMEnable(ws2811TIM);

    ...
    
}
```

blckmn, 2016-07-10:
Another example:

```
typedef struct motor_s {
    IO_t io;
    TIM_t timer;
    bool enabled;
} motor_t;

motor_t motors[MAX_MOTOR_COUNT];

void motorInit(void)
{
    for (uint8_t i; i < MAX_MOTOR_COUNT; i++) {

        motors[i].io = IOByFunction(OWNER_MOTOR, RESOURCE_OUTPUT, RESOURCE_INDEX(i));
        motors[i].timer = TIMERByFunction(OWNER_MOTOR, RESOURCE_OUTPUT, RESOURCE_INDEX(i));

        if (!(motors[i].io && motors[i].timer))
            continue;

        IOConfigGPIO(motors[i].io, IO_CONFIG(GPIO_Speed_50MHz, GPIO_Mode_AF_PP));
        TIMConfigOC(motors[i].timer, ....);

        TIMEnable(motors[i].timer);
        ...
        motors[i].enabled = true;
    }
}
```

savaga, 2016-07-11:
That sounds interesting. So target.c will provide only the list of pairs "IOrec=>TimerRec" with default "owner" (or list of possible owners?). And user can change this assignment to different owner from CLI?

blckmn, 2016-07-11:
Effectively, you are right @savaga : either target.c OR each driver would use the target.h defines for the "default" configuration - i.e. when we setup the config parameter groups we can setup IOInit calls there to initialise the IORecs and TIMRecs on config reset, and config repair.

blckmn, 2016-07-11:
So the defaults are only utilised on reset config or repair config. All other times it is the persisted IORecs or TIMRecs - which are loaded on boot, and manipulated by the resource cli command.

blckmn, 2016-07-11:
Doing it that way - you can actually remove config from the target.h - and simply supply the configuration as part of the reset process for the target through the configurator (i.e. reset information does not exist on the target itself!).

Targets could then become builds for processor versions (F103, F303, F405, F411 etc), and enabled features (GPS, SONAR etc).

martinbudden, 2016-07-11:
This sounds very interesting indeed.

mikeller, 2016-07-11:
> Doing it that way - you can actually remove config from the target.h - and simply supply the configuration as part of the reset process for the target through the configurator (i.e. reset information does not exist on the target itself!).

Not sure if this is the right direction to go - thinking of situations where the Betaflight configurator cannot be used (i.e. on a mobile device), or where the configurator version is not up to date. It would be nice to have the firmware itself as 'self - contained' as possible.

blckmn, 2016-07-11:
@mikeller yes there are pros and cons to both. for the moment my recommendation is to leave it on the target - BUT - there is the possibility if flash space becomes an issue.

martinbudden, 2016-12-01:
@blckmn , can this issue be closed now? I presume it has been superceded.

closed 2016-12-01 by blckmn