betaflight/betaflight #717: 3D mode no throttle response

NightHawk32, 2016-07-10:
Recently did some flights in 3D mode with version 2.8.1, everything worked as expected.
Yesterday I did an update to 2.9.0, but I recognized some strange behaviour:

- No throttle response at all. After arming the motors start spinning and there is response to movement of the quad. But on thottle increase nothing changes, even on negative throttle the output signal does not go below 1500.

- The 3D disable switch does not seem to do anything, even if the AUX channel is in the specified range it does not get highlightet in the configurator.

- Arming should only be possible if the stick is centred and throttle is inside the specified deadband. In 2.9.0 arming is possible regardless of throttle value.

borisbstyle, 2016-07-12:
I will check this

NightHawk32, 2016-08-04:
Did some more testing on this:
Quad flies fine on BF2.8.1, 3D is working as expected. Took this working setup and just flashed it to BF 3.0.0 RC2:

- Still arming possible, regardless of throttle value --> should only be possible when within deadband.
- On arming the motor output jumps from 1500 to 1540 and is locked there. No response to throttle input nor to movements of the quad. Disarming works fine and the output goes back to 1500.

NightHawk32, 2016-08-08:
thanks @borisbstyle for the fast response.

closed 2016-08-08 by NightHawk32