betaflight/betaflight #720: Failsafe Stage 2 not working

rgluckman, 2016-07-11:
When I turn my TX off Failsafe Stage 1 gets triggered and works perfect but no matter what I do Failsafe Stage 2 never engages. I tried making a "Failsafe" mode switch and I still cant get Stage 2 to trigger. With the Failsafe Mode switch engaged the motors just turn off. "failsafe_kill_switch" is also unticked in CF Configurator.
I'm trying to set a specific throttle value so the aircraft comes down slowly instaead of falling to the earth like a rock. Aslo My Crossfire TX Failsafe is set to "Cut"

KiteAnton, 2016-07-11:
Can you supply info on your setup..  ?
Post also output of these commnands in CLI. 

`status`
`tasks`
`dump`

Does the LED on your FC starts to blink with ~1Hz when you try your failsafe ?

borisbstyle, 2016-07-11:
@rgluckman Is your RX failsafe not overruling the FC failsafe perhaps?

rgluckman, 2016-07-11:
The RX Failsafe is set to "Cut". I'm using TBS Crossfire. 
Even when I set up a "Failsafe" switch in the "Modes" tab when this switch is used the Quad just turns off. Stage 2 never kicks in.

rgluckman, 2016-07-11:
KiteAnton- I'm using a TBS Powercube with F60 2450kv motors. TBS Crossfire with TBS Micro RX

rgluckman, 2016-07-11:
# dump

# version
# BetaFlight/COLIBRI_RACE 2.9.0 Jun 25 2016 / 23:48:59 (3d9b180)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO_RATES
feature VBAT
feature RX_SERIAL
feature FAILSAFE
feature ONESHOT125

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB

# map
map AETR1234

# serial
serial 20 1 115200 57600 0 115200
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200

[... 279 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 85
set rc_rate_yaw = 100
set rc_expo = 15
set rc_yaw_expo = 10
set thr_mid = 50
set thr_expo = 0
set roll_rate = 45
set pitch_rate = 45
set yaw_rate = 65
set tpa_rate = 20
set tpa_breakpoint = 1650

#

rgluckman, 2016-07-11:
# status
System Uptime: 9 seconds, Voltage: 9 * 0.1V (1S battery - NOT PRESENT), CPU:4%
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500
Cycle Time: 501, I2C Errors: 5, config size: 1288

rgluckman, 2016-07-11:
# tasks
Task list:
0 - (SYSTEM) max: 4us, avg: 0us, rate: 9hz, total: 1ms
1 - (GYRO/PID) max: 123us, avg: 70us, rate: 1988hz, total: 10sec
2 - (ACCEL) max: 43us, avg: 2us, rate: 994hz, total: 1sec
3 - (ATTITUDE) max: 59us, avg: 28us, rate: 98hz, total: 365ms
4 - (RX) max: 127us, avg: 76us, rate: 49hz, total: 309ms
5 - (SERIAL) max: 9285us, avg: 2us, rate: 99hz, total: 95ms
6 - (BATTERY) max: 11us, avg: 0us, rate: 49hz, total: 10ms
7 - (BEEPER) max: 14us, avg: 0us, rate: 99hz, total: 43ms
14 - (BST_MASTER_PROCESS) max: 7us, avg: 0us, rate: 49hz, total: 15ms

borisbstyle, 2016-09-07:
There is option Land in failsafe page. Try that

closed 2016-09-07 by borisbstyle