betaflight/betaflight #724: Feature Request - Quad "name" in CLI dump/blackbox log?

mike-levenick, 2016-07-11:
First of all, I LOVE the way that BF dumps all sorts of information to the top of a blackbox log now. That's fantastic, and helps a ton with tuning.

How difficult would it be to have a parameter that dumps to the top of a CLI dump and the top of a blackbox log that would be some sort of identifier for a quad? Like a name for the quad.

I have two different quads with the same FC and same version of BF, and I get the CLI dumps and BB logs confused on the two of them quite often. If I could check the first few lines of a CLI dump and see like: 

```
# version
# BetaFlight/COLIBRI_RACE 2.9.0 Jun 25 2016 / 23:48:59 (3d9b180)
# name
# "250 racer"
# dump master
...
```
And to see at the top of a blackbox log:

```
H Firmware type:Cleanflight
H Firmware revision:Betaflight 2.8.1 (c0b00fd) SPRACINGF3
H Firmware date:Jun  8 2016 13:18:36
H name 250 racer
H P interval:1/5
```

martinbudden, 2016-07-11:
Yes, I think this is a good idea.

blckmn, 2016-07-11:
Great idea. Pretty easy to implement too!

mike-levenick, 2016-07-11:
Awesome, glad to hear it. I had a feeling it would be relatively simple to do, but I've never dug into the firmware code to see how variables/parameters are defined, and how difficult it would be to add an echo line to the blackbox/CLI dump.

borisbstyle, 2016-07-13:
This is indeed really easy to implement. Also nice when you get to configure this field in the configurator

KiteAnton, 2016-07-16:
This is now implemented, see #747 and #762

closed 2016-07-16 by KiteAnton