betaflight/betaflight #728: Quads gets out of control, Maybe a FC or ESC Bug?

Teal69, 2016-07-11:
Hi guys, i have a annoying problem with two of my new quads. 
The hardware setup in both quads are : 
First FC we had tried was an Beerotor F3, for later tests we used a Dodo F3 FC , 
Xm30A (Multishot)
EMAX RS2205 2600kv
4s Tattu 75C 1550mAh
DAL 3x5x4.5bn. 
Software we have in used is BF 2.9 and 2.8.1. 
First of all, both of my quads feels very well on the stick since maiden flight and the flight performance is perfect . After some small pitch pumps, the first quad get out of control and didn`t compensate his rotating spin on yaw.  I had test some different looptime/PID settings but nothing work for me.  After that desaster oft the first quad, i tried my second quad which has the  same  setup. Unfortunetly after some small test flieghts, the second quad shows the same issue. After that, i thought that the two beerotors FC are  faulty und bought a dodo fc for testing. With the new Dodo FC and a new Matek PDB, the first quad show the same isssue like the beerotor.
I guess it is an Issue in the FC or ESC Software but i have no idea what i should try for solving my problem.  
  I made a video, which shows the issue after pitchpumping the quad. The blackbox logfile from this flight is also attached. 

 Can someone help me please with my problem? Thanks for all

Dennis 

[image]
[blackbox_log_Bug.TXT](https://github.com/betaflight/betaflight/files/357987/blackbox_log_Bug.TXT)

https://www.youtube.com/watch?v=bZI13UBAzBM

nathantsoi, 2016-07-11:
> Many have reported this issue with DYS XMS30A ESCs.

i run the same escs and fc on my x200 w/ blheli oneshot125, betaflight 3.0
@ 2khz. it flies amazingly well :)

andre92, 2016-07-13:
looks like an issue with motor 2. Betaflight applies max throttle but gyros show the motor is idle. Try switching motor 2 with another motor and see if the problem is with the motor or the esc

borisbstyle, 2016-07-15:
Looks like a motor / ESC issue.

closed 2016-07-15 by borisbstyle