betaflight/betaflight #730: GPS connected and Betaflight freezes after a few minutes

siamect, 2016-07-12:
If I connect my cheapo Banggood Mini F3 Flight Controller 10 DOF to GPS,  betaflight just hangs and no reactions what so ever. If I don't have the GPS connected everything works fine.
I use the SPracingF3 version of betaflight.
My config as follows:
usart 1 empty (USB only)  
USART 2 S-bus X4RSB receiver
USART 3 not connected
Soft serial 1 MW OSD micro 19200 baud
Soft serial 2 GPS VK2828U7G5LF TTL Ublox GPS  19200 baud

Anyone else seen this?

AmiSMB, 2016-07-12:
I put the MW OSD Micro on UART1 with rx and tx connected so I just remove the power to the micro board when I want to use USB connection. Then use UART3 for the X4RSB S-Bus. I am currently moving my S-Port telemetry one of the two smart serial ports so that I can use UART 2 for connecting a GPS. I have not tried it but hopefully will let you know if I find the same tonight.

AmiSMB, 2016-07-12:
You may also want to run the status command under the CLI to see if the CPU is being over used which could also cause this issue.

siamect, 2016-07-12:
I think we have very different hardware because USART 3 is not accessible on mine and the USB to USART chip doesn't have any three state buffer so USART 1 is always controlled by the USB. I checked the circuitry. Annoying because there is a signal meant to control a buffer and it an be made by just one n-channel enhancement mode transistor...  
I will check the CPU usage whenever I get a few minutes left. Thanks for the tip.
What do I do if it is maxed out... increase the cycle time? or are there other tricks?

siamect, 2016-07-13:
This was interesting!!!
I have cpu 11-12% normally and then I connect the GPS and all of a sudden cpu jumps to 100% and stays there forever even if I pull the cable to the GPS...

# tasks
Task list:
0 - (SYSTEM) max: 6us, avg: 0us, rate: 9hz, total: 1ms
1 - (GYRO/PID) max: 311us, avg: 194us, rate: 1968hz, total: 39sec
2 - (ACCEL) max: 148us, avg: 114us, rate: 989hz, total: 11sec
3 - (ATTITUDE) max: 146us, avg: 71us, rate: 98hz, total: 767ms
4 - (RX) max: 159us, avg: 72us, rate: 109hz, total: 993ms
5 - (SERIAL) max: 803us, avg: 2us, rate: 98hz, total: 341ms
6 - (BATTERY) max: 40062us, avg: 1us, rate: 49hz, total: 58ms
7 - (BEEPER) max: 13us, avg: 0us, rate: 98hz, total: 17ms
8 - (GPS) max: 50216us, avg: 410us, rate: 9hz, total: 1sec
9 - (COMPASS) max: 128us, avg: 92us, rate: 9hz, total: 111ms
10 - (BARO) max: 128us, avg: 62us, rate: 7751hz, total: 631ms
12 - (ALTITUDE) max: 54us, avg: 17us, rate: 39hz, total: 110ms

# status
System Uptime: 133 seconds, Voltage: 146 * 0.1V (4S battery - OK), CPU:12%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=BMP280, MAG=HMC5883
Cycle Time: 511, I2C Errors: 2, config size: 1296

here I connect GPS

# status
System Uptime: 138 seconds, Voltage: 146 * 0.1V (4S battery - OK), CPU:100%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=BMP280, MAG=HMC5883
Cycle Time: 1904, I2C Errors: 5677, config size: 1296

# tasks
Task list:
0 - (SYSTEM) max: 7us, avg: 0us, rate: 4hz, total: 2ms
1 - (GYRO/PID) max: 1770us, avg: 967us, rate: 531hz, total: 60sec
2 - (ACCEL) max: 1593us, avg: 882us, rate: 252hz, total: 17sec
3 - (ATTITUDE) max: 146us, avg: 75us, rate: 28hz, total: 1sec
4 - (RX) max: 159us, avg: 81us, rate: 48hz, total: 1sec
5 - (SERIAL) max: 44159us, avg: 2us, rate: 13hz, total: 396ms
6 - (BATTERY) max: 40062us, avg: 1us, rate: 6hz, total: 65ms
7 - (BEEPER) max: 13us, avg: 0us, rate: 13hz, total: 23ms
8 - (GPS) max: 50724us, avg: 5935us, rate: 3hz, total: 2sec
9 - (COMPASS) max: 1357us, avg: 393us, rate: 0hz, total: 159ms
10 - (BARO) max: 1520us, avg: 951us, rate: 531hz, total: 944ms
12 - (ALTITUDE) max: 55us, avg: 18us, rate: 5hz, total: 146ms

siamect, 2016-07-18:
I tested more yesterday evening. 
I have the set gps_auto_config = 1
I run betaflight version 2.9 SPracingF3 
It seems like when the GPS gets more satellites then the CPU goes to 100%. I can get 10 or more satellites showing in the configurator and then no response. The number of satellites are 0 although I can see the signals from many satellites. 
It never show any 3d fix although I have many satellites visible in the configurator and it doesn't show any coordinates. but when I restart the controller then if I go to the GPS menu I see that there are valid coordinates but they disappear after a second or two.

siamect, 2016-07-20:
Ok I found the usart3 and connected the GPS there. No freezing!
So the issue most likely that the softserial cannot handle two streams at the same time?
But I have issue that the 3d fix is going on and off repeatedly...

siamect, 2016-07-22:
But I have issue that the 3d fix is going on and off repeatedly... This was due to my GPS had both NMEA and Ublox messages enabled at the same time. Connect you GPS to a PC. Open U-center (terrible software) an take a look in the message view or whatever it is called. disable all the NMEA messages and then go to config and save...

borisbstyle, 2016-08-02:
I will close this one as the issue seems resovled by not using softserial

siamect, 2016-08-02:
OK no problem, go ahead, but the original bug is still there.

closed 2016-08-19 by borisbstyle