betaflight/betaflight #732: Consider using the gold linker

martinbudden, 2016-07-12:
The `gold` linker is faster than the `ld` linker currently used.

Now that there are so many targets, anything that speeds up the build process is welcome.

digitalentity, 2016-07-12:
@martinbudden I'd like to test `gold` linker as well as gcc-6 in INAV.

martinbudden, 2016-07-12:
Unfortunately, I don't have any experience in setting this up - make files have never really been my forte. @AndersHoglund , perhaps this is something you know about?

AndersHoglund, 2016-07-12:
Makefiles, yes. But compilers and linkers, only use them. Never used gold at all. But sure, I could look into it. Unless @digitalentity not already are doing it.

digitalentity, 2016-07-12:
@AndersHoglund no, I'm not a makefile wizard either. Please have a look if you have time to spare on this :smile:

AndersHoglund, 2016-07-13:
Can't get gcc to accept the "-use-ld=gold" option, seems to be ignored. Did try some other tricks like symlinking "arm-none-eabi-ld" to "/usr/bin/gold", to no effect at all. 

What is needed is to replace the linker command in the Makefile, now "arm-none-eabi-gcc" is used as the linker with the "-o" option and some more $(LD_FLAGS). Problem is that options and parameters are not the same if we where to explicitly use "arm-none-eab-ld" instead. And yet another set of options to use gold... I did try to translate some, but never got it completely right.
Some gnu tool-chain experts need to take over this, I give up.

In those experiments I used gcc 5.4.1 without any hitch. Simply update and go. No problems. But no major advantage either, same build time and slighly larger binaries. 
"make all" takes about 15-16min on Ubuntu 16.04 running in a 4meg Virtualbox on my slow Win10 desktop. Both host and guest OS are 64-bit.

npsm, 2016-07-13:
the linker is most likely not the most time consuming part of the process but compiling is and that won't be solved with changing the linker.

I personally use ccache (https://ccache.samba.org/) it caches build files and re-uses them when possible. In general this speeds up builds by 2x-4x.

To use it, just install ccache as discribed on the website and change this line in your makefile to include ccache:
`CC          = ccache arm-none-eabi-gcc`

martinbudden, 2016-07-14:
@npsm , on my machine it takes about 33 seconds to build a target. Of that time it is about 17 seconds compiling and 16 seconds linking (varies slightly between targets). So speeding up the linker is certainly worth while. I hear reports that `gold` is 5 to 10 times faster than the ordinary linker.

Of course we should use caching as well.

AndersHoglund, 2016-07-14:
Did some tests with ccache. At best a single target (NAZE) was built in about 6s compared to 23s without ccache. A total rebuild of all targets took at best 6m compare to 16m. I.e. 3 times faster.
But results varies. Initial build with empty cache, no speedup (of course). After a "git pull upstream master" I also was back at about 16m. Strange.  A later clean rebuild took 5m40s. 
Will make a PR on ccache.

AndersHoglund, 2016-07-14:
PR #749 issued

npsm, 2016-07-14:
@martinbudden I'm curious what system your building on, I just noticed an odd difference in performance between intel vs AMD (or some other strange reason)
my ~9 year old AMD Athlon X2 does the linking in 1.2 second, so no real reason to optimize here. 
But my 2 year old laptop with Intel i5 does it in 6.9 seconds, it has more RAM more cores a lot higher benchmark score on single and multi-threaded, both have ssd and run the some OS version. Odd.

With this discovery I withdraw my argument that linker optimization isn't needed.

closed 2016-08-19 by borisbstyle