betaflight/betaflight #734: Noisy D on KISS FC

llambkin, 2016-07-12:
Hey Guys,

Just tuning up my KISS FC beast with 3.0.0, and doing some blackbox logging, I would say this is more of a question than an issue.

Is this noise on D normal? I have the filter set to 70 on gyro and 90 on Dterm. Maybe I need to go lower, I just didnt want a sloppycopter. The noise is only when I have throttle on, when I chop the throttle for flips, rolls or decents, it cleans up.

The motors are new DYS MR2205 2300, and the props are HQ5040x3 and the ESC's are KISS

[image]

borisbstyle, 2016-07-13:
Do you hear the noise?

llambkin, 2016-07-13:
Not really, but I swapped my KISS FC back over for a Dodo on the same BF revision, and my PID's didnt translate, and I can run far less filtering without so much noise. They both use the same gyro chip, so I dont know what is going on here. I am no engineer, but could it have something to do with the PWM buffers?

One thing I found hard to do, was to tune in the D, usually you can just wind up D to get rid of propwash osc, but it'd get to a point, and get no better. And looking at the blackbox logs with a High D was horrible.

llambkin, 2016-07-13:
The Dodo certainly sounds cleaner than the KISS FC. I am still yet to FPV it.....

borisbstyle, 2016-07-13:
But your DODO is on an another quad right?

borisbstyle, 2016-07-13:
Can you also post the status and tasks part for that KISS FC?

WaffleFPV, 2016-07-13:
Just thought I would add my two cents, I struggled to tune out the prop wash oscillations no matter what P/D I used, going to the extremes as well. I tried out gyro lowpass of 80 instead but it made it worse. Something is definitely wrong with D term on the Kiss target. I don't have a blackbox but just thought I would say that Liam is not the only one with the issue. Running 2K and 2K. Although I am convinced it is hardware at play, do you have any suggestions for settings i could try out (filter values?) @borisbstyle ?
KISS 24 (OS42), Emax red bottom 2300kv, Dal 5045x3, O-ring vibration damping.

Status: System Uptime: 33 seconds, Voltage: 0 * 0.1V (3S battery - OK), CPU:2%
CPU Clock=72MHz, GYRO=MPU6050
Cycle Time: 504, I2C Errors: 0, config size: 1096

Task list:
00 - (SYSTEM) max: 3us, avg: 0us, rate: 9hz, total: 1ms
01 - (GYRO/PID) max: 240us, avg: 171us, rate: 1992hz, total: 53sec
04 - (RX) max: 107us, avg: 76us, rate: 49hz, total: 697ms
05 - (SERIAL) max: 51449us, avg: 2us, rate: 99hz, total: 356ms
07 - (BEEPER) max: 9us, avg: 0us, rate: 99hz, total: 24ms

llambkin, 2016-07-13:
No not on another quad, I resoldered my Dodo back on the exact same quad the KISS FC came off. Now that it is flying well, I am a bit hesitant to put the KISS back :P. I FPV'd it this afternoon, and OMG it was amazing! Nice work Boris! Rolls and flips stop instantly, I still have some tuning to do on the pitch axis, but its 100% better than the KISS FC.

llambkin, 2016-07-13:
Did you want me to post the blackbox log from my last KISS FC flight?

borisbstyle, 2016-07-13:
@llambkin 
Yes logfile would be great! Preferrably also the log from DODO fc you have on the same quad.
I know the gyros can be also sensitive to input voltage. I am not sure how good the onboard regulator is on KISS FC?

@WaffleFPV 
What dterm filtering values did you try and what D gains? On my usual builds I need around 20ish D to get solid prop wash handling.

This is really hard for me as I dont have KISS target to test with it :(. But might come a long way with the logs. 
I might post some beta builds with better noise rejection to try out.

WaffleFPV, 2016-07-13:
@borisbstyle I used down to 15 and up to 45. 20 is my sweetspot for most also. Seems like little difference in handling no matter what value though. Although arming on the ground with airmode on makes it shake on higher D gains. I increased minthrottle to work around that!
I have tried using a linear regulator, and now I use a 12V pololu which then goes through the Kiss's 5v Reg. I doubt it would be very noisy when it goes through 2 regulators. 

Tomorrow I will try some further extremes and fresh props, and maybe gyrolowpass of down to 50 and see what changes.

llambkin, 2016-07-13:
@borisbstyle, Im in the midst of removing the buffers from my FC, to see if it is that. Maybe it could be the pulses from the PWM output going into the buffers that is upsetting the gyro, but really, if the gyro was upset, we would see that on the gyro traces? Mine are as clean as, on default filtering, with the exception of the noise that makes it back out to the motors.

[LOG00521.TXT](https://github.com/betaflight/betaflight/files/361459/LOG00521.TXT)

llambkin, 2016-07-13:
Here is a log from the Dodo. I dont have one with a 110 filter on the Dterm :(
One thing I noticed with the KISS FC, is that it didnt really matter where the filter was set, the noise was getting through anyway

[LOG00534.TXT](https://github.com/betaflight/betaflight/files/361474/LOG00534.TXT)

llambkin, 2016-07-13:
And here is a log from the KISS FC with 50hz/50hz filtering......it flew pretty bad, but there is less noise there. But it is still there.

[LOG00517.TXT](https://github.com/betaflight/betaflight/files/361485/LOG00517.TXT)

borisbstyle, 2016-07-13:
Did you try 100/50 ?

llambkin, 2016-07-13:
No..... But I did do one on 90/70. I cant remember how it flew, it was only a short flight in the back yard, not FPV there was still sharp noise there, very minimal tho

[LOG00505.TXT](https://github.com/betaflight/betaflight/files/361530/LOG00505.TXT)

llambkin, 2016-07-13:
Is there some code in there that deals with the fact there is buffers there? I have removed the buffers and bridged the solder pads so the PWM pads go directly to the CPU. Now when I power up, all 4 ESC's beep then go out. When I try to recal, it beeps at 2000pwm, but wont do the second beep at the bottom.

borisbstyle, 2016-07-13:
Yes there is a code what deals with that :)

PWM signals are now inverted.

You should have not removed them actually. In theory buffers should only
provide more stable signal to the esc, but now you can flash your escs :)

llambkin, 2016-07-13:
Yeah, I downloaded the workspace last night, and digging through the code, I see that it is inverted. I dont know how to build a hex from the workspace. I can do it from the github, by doing a make TARGET = KISSFC, are you able to build us one with uninverted PWM's? I will use that build until 3.0.0 makes it out to github.

Yeah I was told it would make it more stable. And yeah I figured Id be able to flash BLheli ESCs once they were removed. The next thing to remove in chasing the noise would be the 5v linear bec. WaffleFPV has said he has powered the FC from a 5v switching reg. but to get VBAT you MUST plug the + and - wires into the same power source that powers the 5v linear reg. So technically its still powered and running.

WaffleFPV, 2016-07-13:
12V switching reg -> Vbat -> Kiss FC internal 5V regulator.

llambkin, 2016-07-13:
ok, ahhh, still everything on board the KISS FC is being powered from the linear BEC. 

Also it doesnt matter weather you are powering the FC off 3s or 4s, I have been having the same issue on both......

llambkin, 2016-07-13:
PULEEEEZE Mr @borisbstyle, can you build me a hex to flash with uninverted PWMs. At least it will help me get this thing back up in the air. Weather it helps with the noise issue, I dont know. But the only thing that is different from other boards is those buffers and the linear reg. I am willing to remove that to see if that is inducing the noise also.

martinbudden, 2016-07-13:
@llambkin , you should be able to do this yourself, the change is fairly straightforward:

In `src/main/target/KISSFC/target.c` change all instances of `TIMER_OUTPUT_ENABLED|TIMER_OUTPUT_INVERTED` to `TIMER_OUTPUT_ENABLED` then rebuild.

llambkin, 2016-07-13:
I can do the change, thats easy. I just dont know how to rebuild...... I can do it from the Github, but I dont know how to do it from Jenkins.

Thanks Martin :D

martinbudden, 2016-07-13:
> I tried doing it from the downloaded workspace last night, but make
> complained about it not being from github or something.

Did you do a `git clone` to download the workspace?

llambkin, 2016-07-13:
No I downloaded the zip straight from Jenkins, If I could get the address of the workspace that I can put into the github app, that would make things easy. I have done a git clone on the main Betaflight project, but the KISS FC stuff in there is so old.

Thanks for discussing this with me here @martinbudden. I know this isnt the place for these kind of discussions. So thankyou, I am just letting my inexperience hang out.

-Liam

martinbudden, 2016-07-13:
Well, I'd do a `git clone` of the project and then follow the instructions at `https://github.com/cleanflight/cleanflight/tree/master/docs/development` on how to do a build for your platform. It will probably take a little while to set up, but having it set up will continue to be useful in future.

llambkin, 2016-07-13:
YAY! I got it! Thanks @martinbudden, I had already done a clone, I just needed to press the Sync button on the Master branch to get it up to date. :D
All good, I can now make my own custom build, I will let you know how it goes @borisbstyle.

borisbstyle, 2016-07-13:
@llambkin
Can you do another blackbox on kiss with this option enabled

set debug_mode = cycletime

llambkin, 2016-07-13:
I will put it in the CLI and save it, I will have do test tomorrow morning, its 11:37 here at the moment. And in my area, I cant go outside and test out in the street. :(

Thanks for all the help guys, I will post my log tomorrow morning after my quick flight.

borisbstyle, 2016-07-13:
Somebody showed me this vid and i recognized the name
https://youtu.be/qSWNcFu7A1Q

That doesnt sound and look noisy at all!

Are you perhaps comparing onboard flash log from dodo with external log on
kiss?
External serial logging is much slower and introduces a lot of aliasing as
you dont log all the loops

llambkin, 2016-07-13:
Ok, now that is a relief. I will post the log to that flight below. It is log number 3 that matches up to that video.

This is the log that started this whole "looking for D noise" issue. When I had a look over it, I noticed the very sharp noise on the Dterm, and my logs didnt compare to anything I have seen in Joshua Bardwells videos.

So my lesson is learned, DONT trust blackbox, trust in the way it flies and weather the motors get hot or not? If so, I will return it back to the level as seen in the video, It flew quite well, a few osc here and there, but more tuning, and it will be pretty good.

And Just FYI, I wasnt comparing to the onboard flash log, The dodo's flash log is so tiny, I can barely fit a flight on there even with dumbing down the rate at which it is logged.

So what you are saying is that this "noise" is just the aliasing, and I should expect it in my blackbox logs? Because what I see in the log is the motors looking like they were jittering like crazy, and I was thinking it is bad for the ESC's and motors.

Anyway, I am deeply sorry if I have wasted anyone's time. At least I now have a KISS FC that is capable of flashing BLheli......

Regards -Liam

borisbstyle, 2016-07-15:
@llambkin 
The onboard flash gives more sampling for if you really want to know exactly what is going on.
But your log doesnt look bad. Just some slight aliasing artifacts and not that much noise. And I am talking about the amplitude to the noise. The see saw effect you see cannot be for real. That is always aliasing due to undersampling.

That small amount of noise you see can also be eliminated with the new code I have been testing with, but the noise you experience is totally not something what can upset motors. It does limit you only to how high you can raise your D gain only.

I will provide you a test sample of firmware with better filter for noise reduction if you want to test anyway?

borisbstyle, 2016-07-15:
And yes the motor sound and video are the best noise test and verification :D   

The motors and escs also act as a low pass filter by themselves till certain level!

borisbstyle, 2016-07-17:
Closing this one as it appeared to be non-issue

closed 2016-07-17 by borisbstyle