betaflight/betaflight #739: Oneshot42 Broken after 9 Jul

kc10kevin, 2016-07-13:
I have a user that is using Oneshot42 on the FuryF4 board.  He tried a newer version of 3.0 and it will only initialize one ESC.   To trouble shoot, I sent him custom builds for 2-12 July (last commit from each day).  Works good on 8 July build.  Narrowed it down to commit 085e88c by rolling back that commit and user reports Oneshot42 working.

Sent user a link to this Issue so they can provide input if necessary.

Rossbow, 2016-07-13:
Fury F4 board
Tried oneshot42 and only motor and esc 1 operate, No lights on other 3 esc's, Then tried oneshot125 and it works fine on all esc/motors.
I am using Kiss 24a Esc's , All previous builds work perfect.

Rossbow, 2016-07-13:
It also worked fine on the 9th July build but telemetry on UART3 was broken and has been every build since

sambas, 2016-07-13:
I would try recalibrating the escs. PWM timing has never been correct for F40X in this project before.

borisbstyle, 2016-07-13:
Can you put it on scope and measure the timing?

blckmn, 2016-07-13:
I'll check on scope.

Rossbow, 2016-07-13:
@sambas  I will give that a go, did not think to do that since it was working and had been calibrated on the previous builds,but it makes sense since you mentioned the PWM timing was not correct before but is now.

@borisbstyle   Sorry no scope  ,I Just fly them (:

blckmn, 2016-07-13:
Ok - I think i have found the issue.

https://github.com/betaflight/betaflight/pull/742

When initialised the timer is sitting at 50us, and then comes down to 42us. this might be causing the issue. I would think MS is failing as well, but 125 is fine.

Rossbow, 2016-07-13:
@blckmn   Seems to be working fine now thanks.  Also I made the hacks to the X4R receiver and telemetry was working fine but now on the latest builds it seems broken ?

blckmn, 2016-07-14:
@Rossbow I'll have a look at telemetry again tonight. This issue can be closed

closed 2016-07-14 by blckmn

sambas, 2016-07-14:
@Rossbow no esc recalibration needed?

Rossbow, 2016-07-14:
@sambas   OK so I calibrated and it seemed to fix the problem, but I played with the throttle a little bit and then an esc dropped out, rebooted and I tried again and played with the throttle again and was getting weird tone from the escs on closing the throttle, like a lag on stopping ?
It just did not feel right, So I loaded the new build with the fix and just calibrated and then it seemed fine.
I am yet to fly it because of the telemetry problem and for now I went back to an earlier build where the telemetry is working.