betaflight/betaflight #744: Crash on ZMR250 Size copter - Possible bug?

UltiFix, 2016-07-13:
Specs:
[URL="http://www.johnny-robot.com/skytank250"]Johnny Robot Skytank 250[/URL]
CC3D Running BF 2.9
X4R-SB
[URL="http://www.banggood.com/4-PCS-Eachine-Falcon-250-Flycolor-Fairy-20A-ESC-Electronic-Speed-Controller-p-1041009.html"]3 esc's[/URL] and [URL="http://www.banggood.com/Eachine-Falcon-250-2204-2300KV-CW-CCW-Motor-p-1042955.html"]4 motors[/URL] from eachine falcon 250
1 mrm esc (On motor 3#)
No active braking enabled

So, the craft will be flying fine then it will just death spiral out of the air... This has happend a few times and I got this one on a log at 100%

Here is the link to the log... The reason the log does not go all the way to the end is because my logging is set up on my arm and disarm 
switch...

As the copter is starts its death sprial motor 2 maxes out, why would it do that? I have also calibrated my escs.
[url]http://www.mediafire.com/download/cdo12oz9qv1py93/blackbox_log_2016-07-12_205416.TXT[/url]

Thanks!
-Ulti

andre92, 2016-07-13:
2 things:
- motor 2 is failing, looking at the log I see the copter is spinning with motor 2 at full throttle yet that motor is  falling. Betaflight is correctly increasing the throttle for the falling motor, trying to counter its fall.
- you have a lot of vibration judging from the gyro traces, maybe you need to soft mount the flight controller

UltiFix, 2016-07-13:
Now that you mention it... My copter doesnt descend very fast, its like d is almost to high.. Probably because of the vibrations?

The frame im using is mostly flexible polycarbonate.

Now that you mention it, that motor that got pegged at full throttle didnt want to spin and was rough awhile ago. I just assumed it was some dirt or something that was in there for abit.

Im sure its a bad motor, but if you still think its worth it to switch the motor with a good one I can do it.

andre92, 2016-07-13:
If you are sure the motor is bad, replace it by a new one and don't bother switching it. Vibration measurements in the motors tab will tell you that. Swapping motors only helps if you suspect a faulty esc. A bad motor may be the cause of the vibrations as well. I don't think the D gain is too high, it is just suffering from the noise. With the noise gone, D gain should behave normal.

UltiFix, 2016-07-13:
I have some really weird results... Here is a video showing me rev up each of my motors 1 at a time.

The one that is vibrating alot does spin harder than the others, but I assumed that was because its the one with a different esc.
http://youtu.be/HTpokgq9ku0?hd=1

UltiFix, 2016-07-13:
I took motor 2 apart, cleaned it out, put it back together and did a test flight (Not near other people houses etc, above tall grass for paded landings if it crashes)

It performed just fine, going to check the logs...

UltiFix, 2016-07-13:
Flew 3 more packs through it, no problem... ??

UltiFix, 2016-07-14:
Ok, crashed it and my motor spins a bit then stops, and smells funny. This is motor 2. Time for a new one lol

andre92, 2016-07-14:
I'm afraid the motor has burned out. Could you post a blackbox log from before and after the cleaning? I'm wondering if that helped with the gyro noise.

UltiFix, 2016-07-14:
https://www.mediafire.com/?l2m3luht47u51s5
I dont think it made much of a difference...

KiteAnton, 2016-07-15:
This seems to be due to faulty HW, not betaflight FW related so I close this issue.

closed 2016-07-15 by KiteAnton

andre92, 2016-07-15:
I hope new motor will fix the vibrations, if not, I would mount the controller between rubber rings to keep it noise free. With this amount of noise, the noise is fed back from the gyro into the motors, making it even worse. 
By the way, what are your settings for gyro_lpf, gyro_sync_denom, gyro_lowpass and dterm_lowpass?

UltiFix, 2016-07-15:
Default, on older betaflight when the cleanflight configurator was used I set the I think looptime from 1000 to 500, so none of the auto leveled modes worked but I guess it responded faster