betaflight/betaflight #748: Motolab

oidan, 2016-07-14:


KiteAnton, 2016-07-14:
Have you tried all 4 combinations of alignment? According to target.h for motolab the gyro/acc is aligned 180 degrees compared to the board so if you rotate the board 90 degrees then you should use 270 degrees alignment.

oidan, 2016-07-14:
ah, perhaps I should have tested more before I opened an issue, I will check tonight.

oidan, 2016-07-14:
I can already see this is just a pebkac on my part! looking at the gyro alignment on an image of the tornado i can see that indeed the gyro is 180 rotated compared to the direction arrow :/ 
Rather strange thing to do, it would make sense if the gyro was at an odd angle for some funky board layout but 180 makes no sense. almost like they got the silkscreen wrong then were stuck with it.

oidan, 2016-07-14:
Yup my bad, it's not broken. 
Perhaps this issue could be addressed in the configurator, it would be good to show the default orientation in degrees in brackets next to the "Default" setting.
I was unaware the the gyro/acc already had a 180 offset (why would I?) so i assumed that the alignment setting to turn the board 180 would be "CW 180", I had the acc turned off so i didn't see the issue in the 3d model.

closed 2016-07-14 by oidan

KiteAnton, 2016-07-14:
Good you solved your issue. I know it can be a bit difficult to know the default orientation but there are so many boards with different setup so I am not sure how the configurator could keep track of them all. 
Best thing to make sure it is correct is to test with the acc turned on and make sure the model is moving as expected.