betaflight/betaflight #750: cc3d atom cppm pin 3 ?

yoqyoq, 2016-07-14:
3pin doesnt work in version 2.3.4, and version of 2.9 on 3pin.
i use d4r and cc3d atom.
how i can fix it ?

lichtl, 2016-07-14:
CPPM should be on pin 8

See #361

yoqyoq, 2016-07-15:
I've tried. Not working

KiteAnton, 2016-07-16:
@yoqyoq what is the issue? PPM on CC3E is on pin8 no longer on pin3. If it is not working on pin8 then you must rename the issue, otherwise please close it since not working on pin3 is not an issue.

yoqyoq, 2016-07-18:
I tried on pin3 and pin8. My reomte control didnt pair each other. When i installed application down to 2.3.4 connection to pin3 When i install 2.9 versiyon there is no connection between pin 3 and pin 8

chilly24, 2016-07-18:
Thats not true muste be a config problem.
I fly the cc3d atom with the d8r2 receiver on pin 8 and it works!
Betaflight 2.9 
Is that your first quad? Did you powered up the receiver?

closed 2016-12-19 by borisbstyle