betaflight/betaflight #751: cc3d evo unexpected full throttle bug

BxB00T, 2016-07-14:
I turned my controller on with my armed switch on. It didn't arm the quad, which is good/should be expected, then i flipped the switch to off of arm into fail/safe mode then back to arm and it took off full throttle with throttle at zero

Tomcatt, 2016-07-16:
Your EVO ATOM works?  Mine just locks up.  Are you using the .hex or the .bin file?  I'm test out this CC3D and the REVO out.

blckmn, 2016-07-26:
@BxB00T @Tomcatt is this still an issue? is it reproducible, as I cannot reproduce it.

Tomcatt, 2016-07-31:
CC3D seems fine however my Little Bees aren't.  I can connect but can't read them.  Some sort of error comes up on BLHeli.  I'll have to take a screenshot of it when I get back to the room.

And thanks for fixing the REVO.  It test out great on OS125 using my old KingKong 12A ESC's.  But what would be a good unsynced PWM setting?  If I set it to 4K, the ESC's will reboot at 2000 RPM.

I wanted to set my high end.

closed 2016-11-18 by BxB00T