betaflight/betaflight #757: Quads sounds metal if speeding or accelerating (Hardware issue?)

Teal69, 2016-07-15:
Hi@all,

Im tested my new build quad and notice a nasty noise. This  noise is unsteady and comes often if the quad is speeding or pitch pumping.  I start with logging the flights with blackbox and notice some curious peaks in the Gyro signal of the pitch axis.  What can i do to solve this issue? Is the FC faulty?

Thanks for help :-D

Best wishes 
Dennis
[blackbox_noise.hex.TXT](https://github.com/betaflight/betaflight/files/366187/blackbox_noise.hex.TXT)
[image]

borisbstyle, 2016-07-15:
Looks like MPU6500 issue. Softmount your fc and it might solve it.
Using rubber dampeners on your spacers  can be enough

blckmn, 2016-07-27:
@Teal69 can you please confirm if the soft mounting resolved this issue? If not we can re-open but it seems like an isolated issue related to your hardware combination not an issue with the firmware in general.

closed 2016-07-27 by blckmn

Teal69, 2016-07-27:
Hi my Problem is solved with soft mounting my FC. Thanks for your help