betaflight/betaflight #768: CC3D OPBL build broken.

AndersHoglund, 2016-07-16:
Starting at build 402 the Jenkins build status is RED: http://andwho.sytes.net:8080/job/BorisB_BetaFlight/402/

It is the CC3D_OPBL target that is running out if flash space. At least 2 of the 8 commits in this build causes more flash usage: 

1.     add notch filter configurable with gyro_notch_hz and gyro_notch_q (detail / githubweb)
2.     CF new LED strip port (detail / githubweb)
3.     Fix freeze if LED 0 is color only (detail / githubweb)
4.     MSP to 20 + typo Set MSP to 20 Fix typo in reevaluateLedConfig (detail / githubweb)
5.     visual beeper got left out (detail / githubweb)
6.     Move deadband3d_throttle from MSP_3D to MSP_RC_DEADBAND (detail / githubweb)
7.     Fix white color undefined in led drivers (detail / githubweb)
8.     Tidied whitespace in target.h files (detail / githubweb)

Tried to re-configure CC3D and remove MAG, not  enough still too big. 

Is this the point where we abandon OPBL support and simply remove this target?

borisbstyle, 2016-07-16:
I removed the led strip already. Is it still failing?

AndersHoglund, 2016-07-16:
Oh yeah, still in red: http://andwho.sytes.net:8080/job/BorisB_BetaFlight/404/
Only slightly less overflow

    /usr/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: obj/main/betaflight_CC3D_OPBL.elf section `.data' will not fit in region `FLASH'
    /usr/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 264 bytes

borisbstyle, 2016-07-16:
Than something should be left out from CC3D opbl target to free up some
space

martinbudden, 2016-07-16:
@borisbstyle , I'd like to keep the OPBL build - I use it a lot in testing. For me it is the quickest to reflash.

I have to dash at the moment, but I can sort this out later this afternoon.

borisbstyle, 2016-07-16:
No i mean some feature should be left out from cc3d opbl

AndersHoglund, 2016-07-16:
Turning of MAG helps. Should I PR that ?

AndersHoglund, 2016-07-16:
Removing MAG gives sizes with gcc 5.4.1:

    arm-none-eabi-size ./obj/main/betaflight_CC3D_OPBL.elf
       text    data     bss     dec     hex filename
     114756    1228   11136  127120   1f090 ./obj/main/betaflight_CC3D_OPBL.elf

Sizes with gcc 4.9.3:

    arm-none-eabi-size ./obj/main/betaflight_CC3D_OPBL.elf
       text	   data	    bss	    dec	    hex	filename
     115164	   1224	  11556	 127944	  1f3c8	./obj/main/betaflight_CC3D_OPBL.elf

borisbstyle, 2016-07-16:
Yeah remove it.

AndersHoglund, 2016-07-16:
All Green again. http://andwho.sytes.net:8080/job/BorisB_BetaFlight/405/

closed 2016-07-16 by AndersHoglund