betaflight/betaflight #769: BF 2.9: smartport telemetry occasionally not working

nikybiasion, 2016-07-16:
If i set gyro update and pid loop at the same frequency, smart port telemetry stop working.
i've tried 2k/2k 4k/4k 8k/8k
If i set 4k/2k 2k/1k 8k/4k smartport telemetry works

I use a spracing f3 evo with this port configuration:

serial 20 1 115200 57600 0 115200
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 32 57600 57600 57600 57600

Telemetry working (8k/4k)
# tasks
Task list:
0 - (SYSTEM) max: 2us, avg: 0us, rate: 9hz, total: 0ms
1 - (PID) max: 524us, avg: 274us, rate: 988hz, total: 14sec
- - (GYRO)  rate: 1976hz
4 - (RX) max: 109us, avg: 78us, rate: 48hz, total: 80ms
5 - (SERIAL) max: 283us, avg: 2us, rate: 97hz, total: 12ms
6 - (BATTERY) max: 11us, avg: 0us, rate: 48hz, total: 2ms
7 - (BEEPER) max: 7us, avg: 0us, rate: 98hz, total: 2ms
12 - (TELEMETRY) max: 289us, avg: 3us, rate: 392hz, total: 60ms

# status
System Uptime: 20 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:16%
CPU Clock=72MHz, GYRO=MPU6500
Cycle Time: 130, I2C Errors: 1, config size: 1296

Telemetry not working (4k/4k)
# tasks
Task list:
0 - (SYSTEM) max: 2us, avg: 0us, rate: 9hz, total: 0ms
1 - (GYRO/PID) max: 579us, avg: 90us, rate: 3984hz, total: 17sec
4 - (RX) max: 108us, avg: 77us, rate: 49hz, total: 46ms
5 - (SERIAL) max: 320us, avg: 2us, rate: 99hz, total: 9ms
6 - (BATTERY) max: 11us, avg: 0us, rate: 49hz, total: 0ms
7 - (BEEPER) max: 8us, avg: 0us, rate: 99hz, total: 0ms
12 - (TELEMETRY) max: 287us, avg: 3us, rate: 489hz, total: 8ms

# status
System Uptime: 24 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:7%
CPU Clock=72MHz, GYRO=MPU6500
Cycle Time: 252, I2C Errors: 1, config size: 1296

Telemetry working (4k/2k)
# status
System Uptime: 21 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:6%
CPU Clock=72MHz, GYRO=MPU6500
Cycle Time: 257, I2C Errors: 1, config size: 1296

# tasks
Task list:
0 - (SYSTEM) max: 3us, avg: 0us, rate: 9hz, total: 0ms
1 - (PID) max: 517us, avg: 54us, rate: 1999hz, total: 13sec
- - (GYRO)  rate: 3999hz
4 - (RX) max: 109us, avg: 76us, rate: 49hz, total: 107ms
5 - (SERIAL) max: 1116us, avg: 2us, rate: 99hz, total: 12ms
6 - (BATTERY) max: 11us, avg: 0us, rate: 49hz, total: 3ms
7 - (BEEPER) max: 9us, avg: 0us, rate: 99hz, total: 3ms
12 - (TELEMETRY) max: 289us, avg: 3us, rate: 489hz, total: 45ms

isjfk, 2016-07-30:
Same issue here (maybe worse).

I have a BeeRotor F3 (Basically identical to SPRacing F3) with Baro and Mag ON, but no GPS.
S.Port telemetry on SoftSerial1.

In default loop freq (2K/1K), sometime telemetry work initially when connect battery. But just few seconds later no data received ever. Most of the time it just didn't work.
If I low loop freq to 1K/0.5K, the telemetry work when power on and before arm, but soon it stopped work after arm. After disarm, telemetry resume few seconds later.

I don't think this issue exists on previous v2.8.1.

isjfk, 2016-08-01:
Forget my previous post. It just work at first time. Now it's broken again.

isjfk, 2016-08-01:
I think I find the cause. It's not in the firmware, it's in the configurator.
Open file "msp.js" in configurator, goto line 1459, change following code:
```
                buffer.push(SPECIAL_PARAMETERS.airModeActivateThreshold);
                buffer.push(SPECIAL_PARAMETERS.rcSmoothInterval);
                buffer.push(SPECIAL_PARAMETERS.escDesyncProtection);
```
Into:
```
                buffer.push16(SPECIAL_PARAMETERS.airModeActivateThreshold);
                buffer.push(SPECIAL_PARAMETERS.rcSmoothInterval);
                buffer.push16(SPECIAL_PARAMETERS.escDesyncProtection);
```
Test several times, looks fine now.

isjfk, 2016-08-01:
This defect is fixed already: https://github.com/betaflight/betaflight-configurator/commit/3852cc5beeaff0513945c495e200cbd79c69e146

But current version (v1.3) in Chrome Extension Store still suffer from it.

nikybiasion, 2016-08-01:
I've tried latest configurator, but with gyro update and pid loop at the same frequency, i stil haven't telemetry.
If i use PID integer, telemetry works with 8k/8k or 4k/4k

isjfk, 2016-08-01:
Hi @nikybiasion,
Which version you use? If you use current version in Chrome Store (v1.3) the issue still exists.

You may try this version (v1.6.4), which should have this issue solved:
https://github.com/betaflight/betaflight-configurator/archive/v1.6.4.zip

Install into Chrome following this instruction (the Alternative way):
https://github.com/betaflight/betaflight-configurator#alternative-way

I haven't test v1.6.4, will try it tomorrow.

nikybiasion, 2016-08-01:
i'm using 1.6.4 and i've also tried the latest clone from github.
It works in 2k/1k, but the issue occured in 2k/2k or 4k/4k, when denom is 1

borisbstyle, 2016-08-01:
@nikybiasion 
Thats possible. When you want to run very fast rates the other tasks get less importance to maintain the high speed

nikybiasion, 2016-08-01:
@borisbstyle ok, but telemetry not works at 2k/2k 4k/4k, but works at 4k/2k or 8k/4k when cpu load is higher.
When i use 2.8.0 i have telemetry working with 8k/8k

borisbstyle, 2016-08-01:
But I will consider increasing the task priority for telemetry. That all will hurt performance of course

borisbstyle, 2016-08-01:
@nikybiasion 
Increased the priority of telemetry task for 3.0.0
https://github.com/betaflight/betaflight/commit/50ce9c5be7695c0c02ed466136fbd8918643dd28

nikybiasion, 2016-08-01:
I will try in the next days and i will report back.

nikybiasion, 2016-08-02:
I've tried patching 2.9 and telemetry works with 8k/8k at cost of 35/39% cpu load.
With 3.0.0 cpu load is extremly high, about 80/85%!!

borisbstyle, 2016-08-02:
Yes 3.0 has more math built in the new PID controller :D. 8k/8k is possible
when using board rotation from gyro and not from FC and disable other
things.

But anyway high pid loops didnt show much difference compared to 4k at all
in my tests!

closed 2016-08-19 by borisbstyle