betaflight/betaflight #772: Synced and Unsynced settings appear to operate in inverted sense

MartinHugh, 2016-07-16:
I am scoping the FC ESC 1 pin whilst adjusting Motor 1 in the BF Configurator Motors Tab, and adjusting the synced/unsynced slider switch to observe behavior of OneShot 125.
[image]

When unsynced, cycle time is ~250us @ throttle=2000
when synced, cycle time is ~2.5ms @ throttle=2000

Shouldn't they be the other way around ?

[Note Gyro and PID frequency is 4Khz]

** I should also add that I am using an Alpha version of BF 3.0.0 in order that I can run with X-Racer 3 FC.

blckmn, 2016-07-16:
The issue you are seeing is 100% throttle. PWM at those rates on 125 effectively has no gap between pulses.

Unsynced means not in sync with the pid loop.
Synced means a shot at the end of each pid loop. 

With synced If the pid loop is fast enough there's no dwell time between pulses and you see a saturation of the pwm. Change to OS42, and you'll see better sync with the pid loop.

rav-rav, 2016-07-16:
Seems to be wrongly inverted in the code. https://github.com/betaflight/betaflight/blob/master/src/main/flight/mixer.c#L430

MartinHugh, 2016-07-16:
I am not complaining about the duty cycle. I expect it to be 100% on OS125 at 4Khz.  My point is that it should not be getting synced by the PID loop at all because unsynced is set (so it should be running at 400hz)

blckmn, 2016-07-16:
@MartinHugh sorry - you were absolutely right. Thanks @rav-rav good find.

Fixed in https://github.com/betaflight/betaflight/pull/775

closed 2016-07-16 by blckmn