betaflight/betaflight #774: BF compilation not working in windows

gaelj, 2016-07-16:
Commit 4d69682c9649d48e09ca97c3ec6f23213fdacc82 seems to have broken compilation on my machine :

I get these error messages:

```
Cleaning CC3D succeeded.
make -j ./obj/betaflight_3.0.0_CC3D.hex
make[1] : on entre dans le répertoire « /cygdrive/c/Users/gael/workspace/betaflight »
%% startup_stm32f10x_md_gcc.S
%% adc_stm32f10x.c
%% bus_i2c_stm32f10x.c
arm-none-eabi-gcc.exe: error: /home/gael/.ccache/tmp/adc_stm32f.tmp.cooler-master.7840.i: No such file or directory
arm-none-eabi-gcc.exe: fatal error: no input files
compilation terminated.
Makefile:653 : la recette pour la cible « obj/main/CC3D/drivers/adc_stm32f10x.o » a échouée
make[1]: *** [obj/main/CC3D/drivers/adc_stm32f10x.o] Erreur 1
make[1]: *** Attente des tâches non terminées....
arm-none-eabi-gcc.exe: error: /home/gael/.ccache/tmp/bus_i2c_st.tmp.cooler-master.3364.i: No such file or directory
arm-none-eabi-gcc.exe: fatal error: no input files
compilation terminated.
Makefile:653 : la recette pour la cible « obj/main/CC3D/drivers/bus_i2c_stm32f10x.o » a échouée
make[1]: *** [obj/main/CC3D/drivers/bus_i2c_stm32f10x.o] Erreur 1
make[1] : on quitte le répertoire « /cygdrive/c/Users/gael/workspace/betaflight »
Makefile:727 : la recette pour la cible « hex » a échouée
make: *** [hex] Erreur 2
```

gaelj, 2016-07-16:
Extra info:

- CCACHE variable value is "ccache"
- ccache is available in cygwin

```

gael@cooler-master /cygdrive/c/Users/gael/workspace/betaflight
$ make ccache
RESULT: 0
CCACHE: ccache

gael@cooler-master /cygdrive/c/Users/gael/workspace/betaflight
$ ccache
Usage:
    ccache [options]
    ccache compiler [compiler options]
    compiler [compiler options]          (via symbolic link)

Options:
    -c, --cleanup         delete old files and recalculate size counters
                          (normally not needed as this is done automatically)
    -C, --clear           clear the cache completely
    -F, --max-files=N     set maximum number of files in cache to N (use 0 for
                          no limit)
    -M, --max-size=SIZE   set maximum size of cache to SIZE (use 0 for no
                          limit; available suffixes: G, M and K; default
                          suffix: G)
    -s, --show-stats      show statistics summary
    -z, --zero-stats      zero statistics counters

    -h, --help            print this help text
    -V, --version         print version and copyright information

See also <http://ccache.samba.org>.

```

gaelj, 2016-07-16:
Also, I can confirm the problem disappears if I replace
`CC          := $(CCACHE) arm-none-eabi-gcc`
with
`CC          := arm-none-eabi-gcc`

AndersHoglund, 2016-07-16:
Seems as ccache is broken. File that should have been cached is missing, it seems.
Uninstall ccache and the Makefile will run gcc the old way without ccache. I'll see if I can reproduce, do not have a complete cygwin env setup. Anyone else that can confirm this cygwin problem ?

gaelj, 2016-07-16:
> File that should have been cached is missing

I was just about to confirm this !

gaelj, 2016-07-16:
I have ccache version 3.1.9 in case this could help.

AndersHoglund, 2016-07-16:
That is a bit old, 3.2.4 here.

gaelj, 2016-07-16:
Well...
https://groups.google.com/forum/#!topic/android-ndk/15JI_uaXDug

> you need a non-cygwin version of ccache. That's because the toolchain programs (e.g. arm-linux-androideabi-gcc.exe) do not understand that cygwin paths that ccache is passing to it under the covers (i.e. they can't access /home/zhicai/.ccache/whatever because there is no C:\home\zhicai\.ccache on your machine).

AndersHoglund, 2016-07-16:
There are a lot of bugs fixed in ccache since 3.1.9. For example in 3.1.11 I find this:

    Fixed bug which could result in false cache hits when source code contains '"' followed by " /*" or " //" (with variations). 

Seems you have a false cache hit....

gaelj, 2016-07-16:
Not sure how I can manually update it. I re-installed cygwin only a few weeks ago so this is the latest version it has by default.
Oh well, I guess I'll reinstall cygwin without ccache.

AndersHoglund, 2016-07-16:
Nah, just uninstall ccache.  Or better yet, install Virtualbox or similar with a real Linux distro. Cygwin is a poor substitute...

gaelj, 2016-07-16:
We don't want to make it even more difficult than it already is for developers under windows to get a running setup. The problem won't just be for me...
Cygwin is required for direct compilation from within Eclipse, so even if it's limited, it's very usefull in my case.
A simple fix would be to update this page (CF doc):
https://github.com/cleanflight/cleanflight/blob/946dcd8267169c7e1d6503f1dea038f09022f2d2/docs/development/Building%20in%20Windows.md

AndersHoglund, 2016-07-16:
Well, this is only a problem if ccache is installed. And there is no recommendation to do that, as far as I can see in the linked cygwin guide. And that guide is for CF, that does not even use ccache....

gaelj, 2016-07-16:
I didn't install ccache volontarily.
Updating cygwin and not enabling ccache installation resulted in ccache still being there. Maybe I should have deleted everything and do a fresh install, for the sake of checking this.
Anyway, simply deleting the file `C:\cygwin64\bin\ccache.exe` fixed it.

AndersHoglund, 2016-07-16:
I did not get ccache installed by default, but maybe it is bundled with some other selection. To uninstall you just toggle the selection until it says "uninstall".

AndersHoglund, 2016-07-16:
I just made a new page in the betaflight Wiki, https://github.com/betaflight/betaflight/wiki/Known-Build-Environment-Problems

Would that be OK ?

gaelj, 2016-07-16:
OK ! Thanks ! Closing now.

closed 2016-07-16 by gaelj